rtic-rs/rtic #440: [Book] Add chapter for hardware tasks/interrupts

jhillyerd, 2021-01-17:
The RTIC book would benefit from having an Interrupts/Hardware tasks chapter. As it stands, they aren't introduced formally, so the reader is left scrolling around the Resources and Software tasks chapters to learn about them.

Suggested topics:

- How to define a hardware task
- ` check_interrupt()`, `clear_interrupt_pending_bit()`, `clear_update_interrupt_flag()` still relevant or required in RTIC.

Points to make:

- A task fn can only be bound to single interrupt
- Bound interrupts must be removed from the app(dispatchers) list
- If you are defining a high priority task, you may want to leverage messaging queuing to do larger chunks of work in a lower priority software task

perlindgren, 2021-01-17:
Thanks for the input.

Indeed the text could address the topics/points you mentioned (the info is partially there already, but can be made more explicit and easier to find).

There are also some examples of real applications in the `examples` repository, I think that many users are not aware of them.

GregWoods, 2021-01-18:
>Bound interrupts must be removed from the app(dispatchers) list

Erm.. what? I'd love some elaboration on that. I can't recall reading it anywhere

korken89, 2021-01-18:
@GregWoods The proc-macro will give an error if you have both, as you cannot have an interrupt bound as a hardware task and a dispatcher at the same time.

GregWoods, 2021-01-18:
oops, sorry I deleted my comment after I received additional info from the matrix chat. Good to know you cannot assign to both.

commit f0c31998 (2021-09-25): Merge #528

528: The great 0.6 docs update r=AfoHT a=korken89

Closes #530
Closes #527
Closes #487
Closes #461
Closes #448
Closes #440
Closes #422
PR #528 (2021-09-25): The great 0.6 docs update
Closes #530
Closes #527
Closes #487
Closes #461
Closes #448
Closes #440
Closes #422
closed 2021-09-25 by bors[bot]