rtic-rs/rtic #444: [Book] How to modularize & grow your code w/ 0.6

jhillyerd, 2021-01-26:
It's easy to end up with a very large src/main.rs / `mod app` with RTIC.  Having a next steps section with ideas on how to grow and modularize code.

LU15W1R7H, 2022-02-04:
Yes, I think this is a very important point. Currently I see this as the biggest area where RTIC can improve.
It's very hard to structure a complex RTIC application.

I'm not sure if it was a bit hasty to already make a 1.0 release, because I believe that work in this area would involve a lot of breaking changes...

perlindgren, 2022-02-04:
You can use the `app` just as a binding for entry points using the `extern` keyword. This allows for modularization of the code. 

We are a bit limited by the current procmacro support in Rust. As RTIC needs to perform system-wide analysis all task/resource information needs to be in a  single item (the `mod`).

Food for thought:

You can think of RTIC just as `cortex-m-rt`, but with the (much needed) addition of race- and deadlock-free concurrency in safe user code. 

The message passing, timers etc. are just icing on the cake. One can have other abstraction on top of RTIC, e.g., asynchronous actors or concurrent reactive objects (essentially actors with both synchronous asynchronous communication). 

In this sense RTIC is just the lowest level of abstraction needed for building concurrent applications.

AfoHT, 2022-07-19:
Related to this: Issue #639

eflukx, 2022-07-19:
> The message passing, timers etc. are just icing on the cake. One can have other abstractions on top of RTIC, e.g., asynchronous actors or concurrent reactive objects (essentially actors with both synchronous asynchronous communication).
> 
> In this sense RTIC is just the lowest level of abstraction needed for building concurrent applications.

I would be very interested in the available higher level abstractions available the work nicely along RTIC. For example I could use a pub/sub (or event-broadcast) system that 'unlocks'/starts one or more tasks when some event has happened. Think Semaphores/Eventgroups as available in other RTOSses.

Having a (curated) list of available options (maybe even with some example code! 👍🏻) would be of great help to use RTIC in more advanced scenarios (i.e. real world embedded work).

perlindgren, 2022-07-26:
There is an Actor API PR that provides higher level of abstraction (broadcasting). 

https://github.com/rtic-rs/cortex-m-rtic/pull/537