rtic-rs/rtic #458: schedule behavior in 0.6

MabezDev, 2021-03-23:
`0.5.3` code; the buzzer is correctly scheduled on, then off after a short delay
```rust
let now = now + (5 * TICKS_PER_10MS).cycles();
schedule
    .mmi(now, MmiRequest::Buzzer(Buzzer::On(4_u32.khz().into())))
    .ok();
let off = now + (10 * TICKS_PER_10MS).cycles();
schedule.mmi(off, MmiRequest::Buzzer(Buzzer::Off)).ok();
let now = off + (10 * TICKS_PER_10MS).cycles();
schedule
    .mmi(now, MmiRequest::Buzzer(Buzzer::On(4_u32.khz().into())))
    .ok();
schedule
    .mmi(
        now + (10 * TICKS_PER_10MS).cycles(),
        MmiRequest::Buzzer(Buzzer::Off),
    )
    .ok();
```

`0.6.0` code; every spawn_after after the first spawn_after is spawned instantly (provided there is capacity)
```rust
mmi::spawn(MmiRequest::Buzzer(Buzzer::Off)).ok();
mmi::spawn_after(
    Milliseconds(50u32),
    MmiRequest::Buzzer(Buzzer::On(4_u32.khz().into())),
)
.ok();
mmi::spawn_after(
    Milliseconds(500u32),
    MmiRequest::Buzzer(Buzzer::Off),
)
.ok();
```

Example log with timestamp:

```bash
[2021-03-23T22:55:11.082Z]   INFO  x100/src/mmi.rs:109 - MMI BUZZER ==> Off
[2021-03-23T22:55:15.700Z]   INFO  x100/src/mmi.rs:109 - MMI BUZZER ==> On(Hertz(4000))
[2021-03-23T22:55:15.701Z]   INFO  x100/src/mmi.rs:109 - MMI BUZZER ==> Off
```

The task in question has ample capacity, `32`.

Is there a reason why we can no longer schedule a task to spawn at multiple times in the future?

korken89, 2021-03-24:
Thanks for reporting, this might be a bug.
I'll reproduce and have a look at hacker night tomorrow! :)

korken89, 2021-03-25:
Hi, I have tried to reduce this but am unable to reproduce.
Can you see what we do different? If you want we can try to do this debuging a bit more online in the Matrix chat if you'd like?

The following code produces the following output:

```
14:12:59.928 pre init
14:12:59.928 init
14:12:59.928 Val: 1 at 0 ms
14:12:59.928 idle
14:13:00.927 Val: 2 at 1000 ms
14:13:01.927 Val: 3 at 2000 ms
14:13:02.927 Val: 4 at 3000 ms
14:13:03.928 Val: 5 at 4000 ms
14:13:04.928 Val: 6 at 5000 ms
14:13:05.929 Val: 7 at 6000 ms
14:13:06.929 Val: 8 at 7000 ms
14:13:07.930 Val: 9 at 8000 ms
```

```rust
#![no_main]
#![no_std]

use panic_rtt_target as _;

#[rtic::app(device = stm32l4xx_hal::pac, dispatchers = [USART1, USART2])]
mod app {
    use dwt_systick_monotonic::{
        consts::{U0, U80},
        DwtSystick,
    };
    use rtic::time::duration::*;
    use rtt_target::{rprintln, rtt_init_print};
    use stm32l4xx_hal::prelude::*;

    #[monotonic(binds = SysTick, default = true)]
    type DwtMono = DwtSystick<U80, U0, U0>;

    #[init]
    fn init(cx: init::Context) -> (init::LateResources, init::Monotonics) {
        let mut flash = cx.device.FLASH.constrain();
        let mut rcc = cx.device.RCC.constrain();
        let mut pwr = cx.device.PWR.constrain(&mut rcc.apb1r1);
        let mut dcb = cx.core.DCB;
        let dwt = cx.core.DWT;
        let systick = cx.core.SYST;

        rtt_init_print!(NoBlockSkip, 4096);

        rprintln!("pre init");

        //
        // Initialize the clocks
        //
        let clocks = rcc.cfgr.sysclk(80.mhz()).freeze(&mut flash.acr, &mut pwr);

        // Setup the monotonic timer
        let mono2 = DwtSystick::new(&mut dcb, dwt, systick, clocks.sysclk().0);

        rprintln!("init");

        printer::spawn(1).unwrap();
        printer::spawn_after(Milliseconds(1_000_u32), 2).unwrap();
        printer::spawn_after(Milliseconds(2_000_u32), 3).unwrap();
        printer::spawn_after(Milliseconds(3_000_u32), 4).unwrap();
        printer::spawn_after(Milliseconds(4_000_u32), 5).unwrap();
        printer::spawn_after(Milliseconds(5_000_u32), 6).unwrap();
        printer::spawn_after(Milliseconds(6_000_u32), 7).unwrap();
        printer::spawn_after(Milliseconds(7_000_u32), 8).unwrap();
        printer::spawn_after(Milliseconds(8_000_u32), 9).unwrap();

        // (init::LateResources {}, init::Monotonics(mono2))
        (init::LateResources {}, init::Monotonics(mono2))
    }

    use core::convert::TryInto;

    #[task(capacity = 16)]
    fn printer(_cx: printer::Context, val: u32) {
        let now: Milliseconds = DwtMono::now().duration_since_epoch().try_into().unwrap();
        rprintln!("Val: {} at {} ms", val, now.integer());
    }

    #[idle]
    fn idle(_: idle::Context) -> ! {
        rprintln!("idle");

        loop {
            cortex_m::asm::nop();
        }
    }
}
```

korken89, 2021-03-25:
I am using the latest commit on this branch for testing, can you see if this helps if you use the same? PR https://github.com/rtic-rs/cortex-m-rtic/pull/456

MabezDev, 2021-03-26:
Thanks for looking into this. I tried you minimal example and worked as expected. I think this is actually a difference in the old 0.5.x `ccynt::Monotonic` and the `DwtSystick` monotonic. For what ever reason it seems like the timer queues can't keep up as I am often hitting [this branch](https://github.com/rtic-rs/dwt-systick-monotonic/blob/master/src/lib.rs#L69). Not sure what would cause this as this works fine in 0.5, any ideas?

korken89, 2021-03-26:
Hmm, sound quite weird.
This monotonic should be just as fast as the one in 0.5.

Could you provide more context to help with the search?
E.g. print the value of `now` and the time at which you schedule the task, plus print when the task runs?

MabezDev, 2021-03-26:
So I setup a small printer task that prints every 500ms:

```rust
#[task]
fn printer(_: printer::Context) {
    let now: Milliseconds = MonotonicTimer::now()
        .duration_since_epoch()
        .try_into()
        .unwrap();
    info!("Printer spawned at {:?}", now.integer());
    printer::spawn_after(500_u32.milliseconds()).ok();
}
```
Output
```rust
[2021-03-26T15:53:28.983Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 1317
[2021-03-26T15:53:30.449Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 2782
[2021-03-26T15:53:37.789Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 10120
[2021-03-26T15:53:40.453Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 12784
[2021-03-26T15:53:47.965Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 20290
[2021-03-26T15:53:50.449Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 22785
```
Output when the task is given the highest priority
```rust
[2021-03-26T15:57:14.755Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 0
[2021-03-26T15:57:15.962Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 1202
[2021-03-26T15:57:16.916Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 2160
[2021-03-26T15:57:19.681Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 4929
[2021-03-26T15:57:21.591Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 6825
[2021-03-26T15:57:26.933Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 12169
[2021-03-26T15:57:29.356Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 14594
[2021-03-26T15:57:31.760Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 16993
[2021-03-26T15:57:36.929Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 22170
```

MabezDev, 2021-03-26:
Hmmm... I just removed WFI from my idle loop and everything is now operating normally.

```rust
[2021-03-26T16:12:34.020Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 0
[2021-03-26T16:12:34.513Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 500
[2021-03-26T16:12:35.016Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 1000
[2021-03-26T16:12:35.519Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 1500
[2021-03-26T16:12:36.021Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 2000
[2021-03-26T16:12:36.514Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 2501
[2021-03-26T16:12:37.027Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 3001
[2021-03-26T16:12:37.520Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 3501
[2021-03-26T16:12:38.013Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 4001
[2021-03-26T16:12:38.516Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 4502
[2021-03-26T16:12:39.019Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 5002
[2021-03-26T16:12:39.522Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 5502
[2021-03-26T16:12:40.014Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 6002
[2021-03-26T16:12:40.527Z]   INFO  x100-bin/src/main.rs:1502 - Printer spawned at 6503

```

korken89, 2021-03-26:
Oh. Does you MCU disable Systick or DWT in WFI? 
I have not seen this myself yet.

MabezDev, 2021-03-26:
So I *think* I might have figured out what is going on here.

Firstly it wasn't the`wfi` instruction its self that was the issue, I had a `SLEEP_TIME` resource locking around the wfi call. The other task using `SLEEP_TIME` had a priority of 3.

```rust
c.resources.SLEEP_TIME.lock(|sleep| {
    let before = DWT::get_cycle_count();
    wfi(); /* CPU is idle here waiting for interrupt */
    let after = DWT::get_cycle_count();
    *sleep += after.wrapping_sub(before);
});
```

Now obviously this was working in 0.5.x so I decided to have a look at the expanded output. In particular the SysTick priorities:

0.5.3:
```rust
core.SCB.set_priority(
    rtic::export::SystemHandler::SysTick,
    rtic::export::logical2hw(6u8, crate::hal::stm32::NVIC_PRIO_BITS),
);
```
0.6:
```rust
core.SCB.set_priority(
    rtic::export::SystemHandler::SysTick,
    rtic::export::logical2hw(1u8, crate::hal::stm32::NVIC_PRIO_BITS),
);
```

I believe what is happening is the scheduler is behaving correctly, but the resource lock of priority 3 is masking off the interrupts until a high priority task is triggered, at which point it tries to catch up, hence I kept hitting the in the past branch on `DwtSystick`.

In hind sight, using the `SLEEP_TIME` resource in a high priority task is not a good idea, and I will change that. With that said, I think the SysTick should have a higher priority as it is used to spawn possibly higher priority tasks which could get lost or delayed if SysTick prio is too low. Is there a reason it has changed? Perhaps it can be configured via the `#[monotonic]` attribute?

MabezDev, 2021-03-26:
> Perhaps it can be configured via the #[monotonic] attribute?

Maybe I should have just tried it before commenting :laughing:, that worked!

I've opened https://github.com/rtic-rs/cortex-m-rtic/issues/461 to make sure this gets documented somewhere, but I think my issue is now resolved; thanks again for your help @korken89!

closed 2021-03-26 by MabezDev

korken89, 2021-03-29:
Great that you found it and thanks for the final root cause!

jordens, 2021-10-04:
Just a note for those still hitting this in various circumstances (if e.g. Monotonic is delaying very-high-priority IRQs now but wasn't before) and being confused by the changelog entry saying it now follows 0.5 behavior again.
The behavior in 0.6.0-rc.1 (and later) is different from both 0.5 and 0.6.0-alpha.5 (or earlier):

* 0.5 assigned the highest priority of all **scheduled** tasks
* 0.6.0-alpha.5 assigned the lowest priority (1) by default
* 0.6.0-rc.1 assigns the highest overall priority which may be larger than the highest priority of all scheduled tasks

korken89, 2021-10-04:
Ah right that's true, the change I made is a bit more extreme than in 0.5.

Thanks for commenting!