rtic-rs/rtic #460: unreachable_unchecked in spawn_at

MabezDev, 2021-03-26:
I am getting a UDF hard fault when building the following example - only in debug mode on `rustc 1.50.0 (cb75ad5db 2021-02-10)`, works fine in release:

```rust
#![no_main]
#![no_std]

use cortex_m_rt::exception;
use panic_itm as _;

#[macro_use]
extern crate log;

#[rtic::app(device = stm32f2xx_hal::stm32, dispatchers = [USART1, USART2])]
mod app {
    use dwt_systick_monotonic::{
        consts::{U0, U120},
        DwtSystick,
    };
    use rtic::time::duration::*;
    use stm32f2xx_hal::prelude::*;

    #[monotonic(binds = SysTick, default = true)]
    type DwtMono = DwtSystick<U120, U0, U0>;

    
    pub const LOG_LEVEL: log::LevelFilter = log::LevelFilter::Info;
    pub type LoggerType = ItmLogger<cortex_m_log::printer::itm::ItmAssumeSync>;
    use cortex_m_log::log::Logger as ItmLogger;
    use cortex_m_log::{
        destination::Itm as ItmDestination, log::trick_init, printer::itm::ItmAssumeSync,
    };

    #[init]
    fn init(cx: init::Context) -> (init::LateResources, init::Monotonics) {
        let mut dcb = cx.core.DCB;
        let dwt = cx.core.DWT;
        let systick = cx.core.SYST;

        let rcc = cx.device.RCC.constrain();
        let clocks = rcc
            .cfgr
            .use_hse(25.mhz())
            .sysclk(120_000_000.hz())
            .pclk1(30.mhz())
            .pclk2(60.mhz())
            .use_pll48mhz()
            .lse(32.khz())
            .freeze();

        // Setup the monotonic timer
        let mono2 = DwtSystick::new(&mut dcb, dwt, systick, clocks.sysclk().0);

        // initialize the logging framework
        let logger = ItmLogger {
            inner: unsafe { ItmAssumeSync::new(ItmDestination::new(cx.core.ITM)) },
            level: LOG_LEVEL,
        };

        unsafe {
            static mut LOGGER: MaybeUninit<LoggerType> = MaybeUninit::uninit();
            LOGGER.as_mut_ptr().write(logger);
            let log: &'static mut _ = &mut *LOGGER.as_mut_ptr();
            trick_init(&log).unwrap_or_else(|err| {
                panic!("Failed to get initialize the logger {:?}", err);
            });
        }

        printer::spawn(1).unwrap();
        printer::spawn_after(Milliseconds(1_000_u32), 2).unwrap();
        printer::spawn_after(Milliseconds(2_000_u32), 3).unwrap();
        printer::spawn_after(Milliseconds(3_000_u32), 4).unwrap();
        printer::spawn_after(Milliseconds(4_000_u32), 5).unwrap();
        printer::spawn_after(Milliseconds(5_000_u32), 6).unwrap();
        printer::spawn_after(Milliseconds(6_000_u32), 7).unwrap();
        printer::spawn_after(Milliseconds(7_000_u32), 8).unwrap();
        printer::spawn_after(Milliseconds(8_000_u32), 9).unwrap();

        // (init::LateResources {}, init::Monotonics(mono2))
        (init::LateResources {}, init::Monotonics(mono2))
    }

    use core::{convert::TryInto, mem::MaybeUninit};

    #[task(capacity = 16)]
    fn printer(_cx: printer::Context, val: u32) {
        let now: Milliseconds = DwtMono::now().duration_since_epoch().try_into().unwrap();
        info!("Val: {} at {} ms", val, now.integer());
    }

    #[idle]
    fn idle(_: idle::Context) -> ! {
        info!("idle");

        loop {
            cortex_m::asm::nop();
        }
    }
}

#[exception]
fn HardFault(ef: &cortex_m_rt::ExceptionFrame) -> ! {
    panic!("Hard fault {:?}", ef);
}

```

```rust
[2021-03-26T14:41:18.083Z]   panicked at 'Hard fault ExceptionFrame { r0: 0x00000000, r1: 0x00000000, r2: 0x0
[2021-03-26T14:41:18.084Z]   0000000, r3: 0x00000001, r12: 0x00000001, lr: 0x08004a53, pc: 0x0800031a, xpsr: 
[2021-03-26T14:41:18.084Z]   0x61000000 }', src/main.rs:103:5
```

```asm
08004a20 <rtic_06_testing::app::printer::DwtMono::spawn_at::{{closure}}>:
 8004a20:	b580      	push	{r7, lr}
 8004a22:	466f      	mov	r7, sp
 8004a24:	b08a      	sub	sp, #40	; 0x28
 8004a26:	460b      	mov	r3, r1
 8004a28:	9006      	str	r0, [sp, #24]
 8004a2a:	f88d 101c 	strb.w	r1, [sp, #28]
 8004a2e:	9208      	str	r2, [sp, #32]
 8004a30:	f240 01b4 	movw	r1, #180	; 0xb4
 8004a34:	f2c2 0100 	movt	r1, #8192	; 0x2000
 8004a38:	9002      	str	r0, [sp, #8]
 8004a3a:	4608      	mov	r0, r1
 8004a3c:	9301      	str	r3, [sp, #4]
 8004a3e:	f7fe fe93 	bl	8003768 <core::option::Option<T>::as_mut>
 8004a42:	9003      	str	r0, [sp, #12]
 8004a44:	e7ff      	b.n	8004a46 <rtic_06_testing::app::printer::DwtMono::spawn_at::{{closure}}+0x26>
 8004a46:	9803      	ldr	r0, [sp, #12]
 8004a48:	2800      	cmp	r0, #0
 8004a4a:	d103      	bne.n	8004a54 <rtic_06_testing::app::printer::DwtMono::spawn_at::{{closure}}+0x34>
 8004a4c:	e7ff      	b.n	8004a4e <rtic_06_testing::app::printer::DwtMono::spawn_at::{{closure}}+0x2e>
 8004a4e:	f7fb fc64 	bl	800031a <core::hint::unreachable_unchecked>
 8004a52:	defe      	udf	#254	; 0xfe
 8004a54:	9803      	ldr	r0, [sp, #12]
 8004a56:	9009      	str	r0, [sp, #36]	; 0x24
 8004a58:	f240 0130 	movw	r1, #48	; 0x30
 8004a5c:	f2c2 0100 	movt	r1, #8192	; 0x2000
 8004a60:	9000      	str	r0, [sp, #0]
 8004a62:	4608      	mov	r0, r1
 8004a64:	9902      	ldr	r1, [sp, #8]
 8004a66:	9a01      	ldr	r2, [sp, #4]
 8004a68:	9b00      	ldr	r3, [sp, #0]
 8004a6a:	f7ff f8bc 	bl	8003be6 <rtic::tq::TimerQueue<Mono,Task,N>::enqueue_unchecked>
 8004a6e:	e7ff      	b.n	8004a70 <rtic_06_testing::app::printer::DwtMono::spawn_at::{{closure}}+0x50>
 8004a70:	b00a      	add	sp, #40	; 0x28
 8004a72:	bd80      	pop	{r7, pc}
```
deps:
```toml
[dependencies]
cortex-m = "0.7.2"
cortex-m-log = {version = "0.6.2", features = ["itm", "log-integration"]}
cortex-m-rt = "0.6.13"
cortex-m-rtic = {version = "=0.6.0-alpha.1"}
dwt-systick-monotonic = "=0.1.0-alpha.0"
panic-itm = "0.4.1"
log = "0.4.6"
```

korken89, 2021-04-08:
Thanks for reporting! I am not sure why this is happening, but I'll start finding the root cause.

korken89, 2021-04-08:
This is now fixed in #466

PR #466 (2021-04-08): Fix for type aliases in `mod app`, UB in `spawn_at`, and `#[cfg]` in hardware tasks
Type aliases such as the following did not work in `0.6-alpha`:

```rust
use rtic::app;

#[app(device = lm3s6965, dispatchers = [SSI0])]
mod app {
    type Test = u32;

    #[task]
    fn t1(_: t1::Context, _val: Test) {}
}
```

Plus that accessing associated constants of monotonic timers was not working as it should dues to the syntax and codegen transforming:

```rust
    #[monotonic(binds = SysTick, default = true)]
    type MyMono = DwtSystick<8_000_000>; // 8 MHz
```

into

```rust
    mod MyMono {
        [... cut]
commit 83cdf00e (2021-04-08): Merge #466

466: Fix for type aliases in `mod app`, UB in `spawn_at`, and `#[cfg]` in hardware tasks r=AfoHT a=korken89

Type aliases such as the following did not work in `0.6-alpha`:

```rust
use rtic::app;

#[app(device = lm3s6965, dispatchers = [SSI0])]
mod app {
    type Test = u32;

    #[task]
    fn t1(_: t1::Context, _val: Test) {}
}
```

Plus that accessing associated constants of monotonic timers was not working as it should dues to the syntax and codegen transforming:

```rust
    #[monotonic(binds = SysTick, default = true)]
    type MyMono = DwtSystick<8_000_000>; // 8 MHz
```

i [... cut]
closed 2021-04-08 by bors[bot]