rtic-rs/rtic #461: Add 0.6 migration note about `Monotonic`

MabezDev, 2021-03-26:
I think it's worth noting that the default timer queue priority is now `1`, unless specified. I have a feeling many people will run into https://github.com/rtic-rs/cortex-m-rtic/issues/458 without this information.

Perhaps an example setting the `Monotonic` priority would be a good idea too.

korken89, 2021-03-29:
Hi, thanks for the feedback!
I fully agree, I'm also not sure that following the same default of tasks makes sense for the monotonic. 
Or just not fixing the priority a default making users aware of the choice?

MabezDev, 2021-03-29:
Perhaps a "sane" default with the ability to customize? I'm not sure how the prio of 6 was assigned to SysTick in 0.5.x in my project,  but that same logic could be used to generate a reasonable default priority.

korken89, 2021-03-29:
Unfortunately the logic from 0.5 does on work in 0.6.
In 0.5 we took the maximum priority of all schedulable tasks, which were given by the `scheudle` parameter in the macro.
In spawn/schedule form anywhere this information is no longer available.

korken89, 2021-09-22:
This is with #528  and #529 close to the old behavior of 0.5

commit f0c31998 (2021-09-25): Merge #528

528: The great 0.6 docs update r=AfoHT a=korken89

Closes #530
Closes #527
Closes #487
Closes #461
Closes #448
Closes #440
Closes #422
PR #528 (2021-09-25): The great 0.6 docs update
Closes #530
Closes #527
Closes #487
Closes #461
Closes #448
Closes #440
Closes #422
closed 2021-09-25 by bors[bot]