rtic-rs/rtic #462: Attributes for hardware tasks

jordens, 2021-03-29:
I would like to place a hardware task in ITCM. With RTIC 0.5 it mostly works fine if I place my task with `#[link_section = ...]`. However, this generates one wrapper function (the actual ISR) from RTIC that is not in ITCM and another `LongThunk` that is also not in ITCM. If I could assign attributes to the RTIC ISR (below for my reference) calling my hardware task, I'd expect both stubs/wrappers to disappear through inlining.

https://github.com/rtic-rs/cortex-m-rtic/blob/3c86d713a6f8fdb052de80380a17468090e42624/macros/src/codegen/hardware_tasks.rs#L43-L56)

perlindgren, 2021-03-29:
Hi jordens.
Thanks for the input!

I think this part has been a bit slept on, and proper tests is lacking in RTIC 0.5. This is something that is on the "TODO" list for RTIC 0.6. 
/Per

closed 2021-04-08 by bors[bot]

jordens, 2021-04-08:
Looks like this works now. Thanks!

PR #466 (2021-04-08): Fix for type aliases in `mod app`, UB in `spawn_at`, and `#[cfg]` in hardware tasks
Type aliases such as the following did not work in `0.6-alpha`:

```rust
use rtic::app;

#[app(device = lm3s6965, dispatchers = [SSI0])]
mod app {
    type Test = u32;

    #[task]
    fn t1(_: t1::Context, _val: Test) {}
}
```

Plus that accessing associated constants of monotonic timers was not working as it should dues to the syntax and codegen transforming:

```rust
    #[monotonic(binds = SysTick, default = true)]
    type MyMono = DwtSystick<8_000_000>; // 8 MHz
```

into

```rust
    mod MyMono {
        [... cut]
commit 83cdf00e (2021-04-08): Merge #466

466: Fix for type aliases in `mod app`, UB in `spawn_at`, and `#[cfg]` in hardware tasks r=AfoHT a=korken89

Type aliases such as the following did not work in `0.6-alpha`:

```rust
use rtic::app;

#[app(device = lm3s6965, dispatchers = [SSI0])]
mod app {
    type Test = u32;

    #[task]
    fn t1(_: t1::Context, _val: Test) {}
}
```

Plus that accessing associated constants of monotonic timers was not working as it should dues to the syntax and codegen transforming:

```rust
    #[monotonic(binds = SysTick, default = true)]
    type MyMono = DwtSystick<8_000_000>; // 8 MHz
```

i [... cut]