rtic-rs/rtic #470: Const generic scoping with monotonic macro

mciantyre, 2021-04-11:
I'm looking at the [`schedule.rs` example](https://github.com/rtic-rs/cortex-m-rtic/blob/43c5ad79c27fbdefa00e2373eba554ec11e1d9df/examples/schedule.rs#L11-L18)'s monotonic type alias, simplified below.

```rust
#[rtic::app(device = lm3s6965, dispatchers = [SSI0])]
mod app {
    use dwt_systick_monotonic::DwtSystick;

    #[monotonic(binds = SysTick, default = true)]
    type MyMono = DwtSystick<8_000_000>; // 8 MHz

    // ...
}
```

This builds as expected. But suppose I want to move `8_000_000` to a `const`, making it easy to reference in both the const argument, and in [the `DwtSystick` constructor](https://github.com/rtic-rs/cortex-m-rtic/blob/43c5ad79c27fbdefa00e2373eba554ec11e1d9df/examples/schedule.rs#L26):

```rust
#[rtic::app(device = lm3s6965, dispatchers = [SSI0])]
mod app {
    use dwt_systick_monotonic::DwtSystick;

    const HZ: u32 = 8_000_000; // 8 MHz

    #[monotonic(binds = SysTick, default = true)]
    type MyMono = DwtSystick<HZ>; 

    // ...
}
```

Building reveals a sequence of errors. The compiler's hints are helpful, and we can make it work with a fully-qualified, brace-delimited constant:

```rust
#[monotonic(binds = SysTick, default = true)]
type MyMono = DwtSystick<{ crate::app::HZ }>; 
```

However, I was anticipating that it could work something like [this](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=f019217823f93bffa817e3069e7305e4). We can skip the fully-qualified path, because the type alias and constant are in the same scope.

Could RTIC support this UX? Though its unexpected, requiring the complete path isn't too troublesome, particularly when the compiler guides us to the fix.

- rustc version 1.51.0
- cortex-m-rt version 0.6.0-alpha.2

korken89, 2021-04-13:
Thanks for reporting!

For a quick look this looks like a codegen bug, I'll investigate.

MabezDev, 2021-04-15:
I have this working, provided the `const` is defined outside the `mod app`. There seems to bit a bit of odd behavior around the monotonic attribute, for example user `super::*;` inside `mod app` yields this error:

```rust
note: `MonotonicTimer1` could refer to the module defined here
   --> x100-bin/src/main.rs:62:1
    |
62  | #[rtic::app(device = crate::hal::stm32, peripherals = true, dispatchers = [CAN1_TX, CAN1_RX0, CAN1_RX1, CAN1_SCE])]
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
note: `MonotonicTimer1` could also refer to the type alias imported here
   --> x100-bin/src/main.rs:137:9
    |
137 |     use super::*;
    |         ^^^^^^^^
    = help: consider adding an explicit import of `MonotonicTimer1` to disambiguate
```

I hope that helps with the debugging, @korken89 :)

mciantyre, 2021-05-29:
Fixed in #485, and available in 0.6.0-alpha.4.

closed 2021-05-29 by mciantyre