rtic-rs/rtic #472: New monotonic spawn_after() causes fmt bloat

jhillyerd, 2021-04-15:
When adding my first `spawn_after(Milliseconds(500_u32))` call using the new monotonic, my .TEXT section went from 40.1KiB to slightly over 64KiB. 

The worst offenders are all formatting related:

```
File  .text    Size           Crate Name
0.1%   9.6%  6.2KiB             std core::num::flt2dec::strategy::dragon::format_shortest
0.1%   8.2%  5.3KiB             std core::num::flt2dec::strategy::dragon::format_exact
0.0%   3.8%  2.4KiB             std core::num::flt2dec::strategy::grisu::format_shortest_opt
0.0%   1.9%  1.2KiB             std core::num::flt2dec::strategy::grisu::format_exact_opt
0.0%   1.6%  1.0KiB             std core::fmt::float::float_to_decimal_common_exact
0.0%   1.4%    932B             std compiler_builtins::int::specialized_div_rem::u64_div_rem
0.0%   1.2%    824B             std core::num::bignum::Big32x40::mul_digits
0.0%   1.2%    798B             std core::fmt::float::float_to_decimal_common_shortest
0.0%   1.0%    644B             std core::fmt::Formatter::write_formatted_parts
0.0%   1.0%    642B             std core::num::flt2dec::strategy::dragon::mul_pow1
```

My project is running on a stm32f103 with 64KiB of flash, I have been avoiding std::fmt as it doesn't fit.

embedded-time Durations implement fmt::Display, I assume this is the source of the bloat: https://docs.rs/embedded-time/0.10.1/src/embedded_time/duration.rs.html#513

I don't think implementing my own Milliseconds type is an option, due to private types: https://github.com/FluenTech/embedded-time/issues/98

https://docs.rs/embedded-time/0.10.1/embedded_time/duration/struct.Generic.html may be a possible work-around, I have yet to try it.

mattico, 2021-04-15:
rustc should be passing `--gc-sections` or equivalent to the linker since `-C link-dead-code` is disabled by default. Could try `-C link-arg=--trace-symbol=<symbol_name>` to see what's referencing the formatting code, might need to use the unmangled symbol name.

jhillyerd, 2021-04-17:
My cargo profile:

```
# Debug info does not occupy flash
[profile.release]
codegen-units = 1
debug = 2
debug-assertions = false
incremental = false
lto = "fat"
opt-level = "z"
overflow-checks = false
```

Final build step in release mode:

`rustc --crate-name hw_cpu_firmware --edition=2018 src\main.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C opt-level=z -C lto=fat -C codegen-units=1 -C debuginfo=2 --cfg "feature=\"default\"" --cfg "feature=\"defmt-default\"" -C metadata=7d39dfed40ffca07 -C extra-filename=-7d39dfed40ffca07 --out-dir C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps --target thumbv7m-none-eabi -L dependency=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps -L dependency=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\release\deps --extern cortex_m=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libcortex_m-1b80b3f15d2a2aa4.rlib --extern rtic=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\librtic-87d5c60eb76928c4.rlib --extern defmt=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libdefmt-f76272be7e8b60ce.rlib --extern defmt_rtt=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libdefmt_rtt-a1235029dd1a3b33.rlib --extern dwt_systick_monotonic=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libdwt_systick_monotonic-7691bde95e64caad.rlib --extern embedded_graphics=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libembedded_graphics-762285988397bfc3.rlib --extern embedded_hal=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libembedded_hal-9a17dca37f65301d.rlib --extern heapless=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libheapless-ab59549aef9e67be.rlib --extern mutex_trait=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libmutex_trait-03772ba3a590e1e8.rlib --extern panic_probe=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libpanic_probe-799310700ea65443.rlib --extern postcard=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libpostcard-7e93d5b468aa67cf.rlib --extern rtic_core=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\librtic_core-6cac66af11905036.rlib --extern rtic_monotonic=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\librtic_monotonic-5590a1a9f9e8cb8d.rlib --extern shared=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libshared-8b67ac31b55065ef.rlib --extern ssd1306=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libssd1306-419c6bd137e3cd41.rlib --extern stm32_usbd=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libstm32_usbd-04da193fe98d1fa9.rlib --extern stm32f1xx_hal=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libstm32f1xx_hal-c69b1f864958f8da.rlib --extern usb_device=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libusb_device-d16dc35474996bbb.rlib --extern usbd_serial=C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\deps\libusbd_serial-7b409d9cbe0818e2.rlib -C link-arg=-Tlink.x -C link-arg=-Tdefmt.x -C link-arg=--nmagic -L C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\build\cortex-m-66e155e87183711a\out -L C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\build\defmt-df7a610c3d1e3538\out -L C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\build\cortex-m-rt-514bcfecf1078800\out -L C:\Users\james\Documents\GitHub\hw-cpu\firmware\target\thumbv7m-none-eabi\release\build\stm32f1-227deca9c6070030\out`

I don't know how to get rustc to print out what it's passing to the linker.

jhillyerd, 2021-04-17:
I tried adding `"-C", "link-arg=--trace-symbol=core::fmt::Formatter::write_formatted_parts",` to .cargo/config, but I don't see any output, not sure if that's because I need to pass in something different, or it's being swallowed somewhere in the build process.

korken89, 2021-04-18:
I will have a look at this. I have a few ideas from where it might be coming.

korken89, 2021-04-18:
@jhillyerd Do you have the project available so I can use it to benchmark?

jhillyerd, 2021-04-18:
Yes, https://github.com/jhillyerd/hw-cpu/compare/rtic060alpha2...alpha2timer -- firmware directory

korken89, 2021-06-02:
There is now a new version available of embedded-time with the fixes.
I'll update RTIC for the next alpha to include this.

korken89, 2021-08-19:
This has been released

closed 2021-08-19 by korken89