rtic-rs/rtic #490: compiling for different devices

pdgilbert, 2021-06-16:
(Somewhat related to #387 and #197.  Also, I'm new to may parts of this so may be missing something obvious. Simple explanations appreciated.)

I have been trying to construct examples that I can run with different device crates. The following example blinks the (onboard) LED with a short pulse very second and a longer blink every ten seconds. Two processes are scheduled, `one` for pulse and `ten` for the longer blink. These spawn a `blink` process that turns the led on and schedules another process to turn it off. 

With minor modifications I can compile and run on `bluepill` and `blackpill stm32f411`. (See BEWARE below before flashing this.)

<details>
<summary>Click to expand code</summary>

```
//! Blink (onboard) LED with short pulse very second and longer blink every ten seconds.
//! Two processes are scheduled, `one` for pulse and `ten` for longer blink. These spawn 
//! a `blink` process that turns the led on and schedules another process to turn it off.

#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use embedded_hal::digital::v2::OutputPin;

use rtic::app;
use rtic::cyccnt::U32Ext;

pub trait LED {
    // depending on board wiring, on may be set_high or set_low, with off also reversed
    // implementation should deal with this difference
    fn on(&mut self) -> ();
    fn off(&mut self) -> ();
}

#[cfg(feature = "stm32f1xx")]
const MILLISECOND :  u32 = 8_000;  // Duration of a millisecond in cyccnt (syst ticks)

#[cfg(feature = "stm32f4xx")]
const MILLISECOND :  u32 = 16_000;  // Duration of a millisecond in cyccnt (syst ticks)

const PULSE:  u32 = 1_000 * MILLISECOND;   // 1 second
const PERIOD: u32 = 10 * PULSE; // 10 seconds

#[cfg(feature = "stm32f1xx")]
use stm32f1xx_hal::{
    gpio::{Output, PushPull, State,
        gpioc::PC13,
    },
    pac,
    pac::{Peripherals}, 
    prelude::*,
};

#[cfg(feature = "stm32f1xx")]
type LedType  =  PC13<Output<PushPull>>;

#[cfg(feature = "stm32f1xx")]
fn setup(dp: Peripherals) -> LedType {

     let mut rcc = dp.RCC.constrain();

     let mut gpioc = dp.GPIOC.split(&mut rcc.apb2);
     let mut led = gpioc.pc13.into_push_pull_output_with_state(&mut gpioc.crh, State::Low);
     //let mut led = gpioc.pc13.into_push_pull_output(&mut gpioc.crh);

     impl LED for PC13<Output<PushPull>> {
             fn on(&mut self) -> () {
             self.set_low().unwrap()
             }
             fn off(&mut self) -> () {
                 self.set_high().unwrap()
             }
     }
     led.off();

     led
}

#[cfg(feature = "stm32f4xx")]
use stm32f4xx_hal::{
    gpio::{Output, PushPull,
        gpioc::PC13,
    },
    pac,
    pac::{Peripherals,},
    prelude::*,
};

#[cfg(feature = "stm32f4xx")]
type LedType  =  PC13<Output<PushPull>>;

#[cfg(feature = "stm32f4xx")]
fn setup(dp: Peripherals) -> LedType {

     let gpioc = dp.GPIOC.split();
     let led = gpioc.pc13.into_push_pull_output();  

     impl LED for PC13<Output<PushPull>> {
             fn on(&mut self) -> () {
             self.set_low().unwrap()
             }
             fn off(&mut self) -> () {
                 self.set_high().unwrap()
             }
     }

     led
}

//  NEED TO SPECIFY DEVICE HERE FOR DIFFERENT HALs

#[app(device = stm32f1xx_hal::pac, peripherals = true, monotonic = rtic::cyccnt::CYCCNT)]
//#[app(device = stm32f4xx_hal::pac, peripherals = true, monotonic = rtic::cyccnt::CYCCNT)]

const APP: () = {
    struct Resources {
        led: LedType,
    }

    #[init(schedule = [one, ten])]
    fn init(cx: init::Context) -> init::LateResources {
        
        //rtt_init_print!();
        //rprintln!("blink example");

        let mut core = cx.core;
        core.DCB.enable_trace();
        cortex_m::peripheral::DWT::unlock(); // needed on some devices
        core.DWT.enable_cycle_counter();

        let device: Peripherals = cx.device;

        let mut led = setup(device);
        
        led.off();
        
        cx.schedule.ten(cx.start + PERIOD.cycles()).unwrap();
        // offset 700 ms so blinks do not overlap
        cx.schedule.one(cx.start + (PULSE - 700 * MILLISECOND).cycles()).unwrap();

        init::LateResources {
            led,
        }
    }

    #[task(resources = [led], spawn = [blink], schedule = [one] )]
    fn one(cx: one::Context) {
       cx.spawn.blink(5u32).unwrap();
       cx.schedule.one(cx.scheduled + PULSE.cycles()).unwrap();
    }

    #[task(resources = [led], spawn = [blink], schedule = [ten] )]
    fn ten(cx: ten::Context) {
       cx.spawn.blink(500u32).unwrap();        
       cx.schedule.ten(cx.scheduled + PERIOD.cycles()).unwrap();
    }

    #[task(priority = 3, capacity = 3, resources = [led] )]
    fn led_on(cx: led_on::Context) {
       cx.resources.led.on();
    }

    #[task(priority = 3, capacity = 3, resources = [led] )]
    fn led_off(cx: led_off::Context) {
       cx.resources.led.off();
    }

    #[task(priority = 2, capacity = 3 , spawn = [led_on], schedule = [led_off] )]
    fn blink(cx: blink::Context, time : u32) {  //time in ms
       cx.schedule.led_off(cx.scheduled + (time * MILLISECOND).cycles()).unwrap();
       cx.spawn.led_on().unwrap();
    }

    //#[cfg(feature = "stm32f1xx")]   CFG DOES NOT WORK HERE
    extern "C" {
        fn TIM2();
        fn TIM3();
        fn TIM4();
        fn QEI0();
    }

//    #[cfg(feature = "stm32f4xx")]CFG DOES NOT WORK HERE
//    extern "C" {
//        fn EXTI0();
//        fn SDIO();
//        fn TIM1();
//        fn TIM2();
//        fn TIM3();
//        fn QEI0();
//    }
};
```
</details>

The minor modifications are to change the `#[app ...` to indicate the device `stm32f1xx_hal::pac` or `stm32f4xx_hal::pac` and the section following `extern "C" {` to indicate the device's free interrupts.  All my attempts to conditionally compile these using `#[cfg(...` attributes have failed. 

Being able to do this would greatly simplify CI testing for multiple device crates. I would like to be able to use 
```
#[cfg(feature = "stm32f4xx")]
#[app(device = stm32f4xx_hal::pac, peripherals = true, monotonic = rtic::cyccnt::CYCCNT)]
```
or

```
#[cfg(feature = "stm32f4xx")]
use stm32f4xx_hal::pac;
 
#[app(device = pac, peripherals = true, monotonic = rtic::cyccnt::CYCCNT)]
```

Is this possible?

BEWARE that flashing this code will mean the processor is often in sleep/low power mode  and there will be a problem reconnecting the probe, giving an error message
`Error: jtag status contains invalid mode value - communication failure`. Among other places, this is discussed at https://github.com/japaric/f3/issues/84. For a bluepill fix (hold reset while loading openocd) see https://github.com/TeXitoi/blue-pill-quickstart.
This is also discussed for probe.rs at https://github.com/probe-rs/probe-rs/issues/350.

perlindgren, 2021-06-22:
RTIC is implemented as a procedural macro. The order of expansion does not allow RTIC to see the expanded results after `cfg` attributes, and its not possible (to my knowledge) to force expansion from inside of the procedural macro. So our hands are a bit tied to this end, which makes it hard to properly handle `cfg` attributes in RTIC applications.

pdgilbert, 2021-06-23:
Does that apply also when the `cfg` is a step removed, as in 
```
#[cfg(feature = "stm32f4xx")]
use stm32f4xx_hal::pac;
 
#[app(device = pac, peripherals = true, monotonic = rtic::cyccnt::CYCCNT)]
```
rather than
```
#[app(device = stm32f4xx_hal::pac,   peripherals = true, monotonic = rtic::cyccnt::CYCCNT)]
```
Or would it be possible to add a helper attribute to `app` to take the place of `cfg`?

little-arhat, 2021-07-01:
@pdgilbert if your app uses different devices in the same way you can move cfg-dispatch into separate module and present `mydevice` to rtic, like here https://github.com/copterust/fcfs-rtfm/blob/master/src/boards.rs#L291 + https://github.com/copterust/fcfs-rtfm/blob/master/src/main.rs#L51

This works for almost everything, except interrupts, as described in #197, however there is a way to fix that:  https://github.com/rtic-rs/cortex-m-rtic/issues/197#issuecomment-784546762. 

I'm not sure if rtic is interested in supporting such case, although it's clearly needed (dev/production board, etc.)

perlindgren, 2021-07-01:
Thanks for the the input. IMHO it would be great to offer some sort of configurable bindings to different platforms, we will have a closer look at this approach.

pdgilbert, 2021-07-01:
@little-arhat thanks for these pointers. I will try to work through this. Perhaps by the time I understand it @perlindgren and others will have figured out  a preferred common way to do it.

My use case is simply that while learning by working through examples I have gathered them with a `setup()` for different platforms [here](https://github.com/pdgilbert/rust-integration-testing). This is partly to satisfy my own curiosity about the extent to which `embedded-hal` delivers on the hardware abstraction, and partly because I think this kind of CI testing can eventually be a helpful resource for moving forward. The [CI is here](https://github.com/pdgilbert/rust-integration-testing/actions).

My hope is to add `rtic` examples but that depends on being able to configure for different platforms.

datdenkikniet, 2021-08-19:
Would using `cfg_attr` be able to help with/solve this issue?

i.e.
```rust
#[cfg_attr(feature = "some_stm", app(..))]
```

datdenkikniet, 2021-08-19:
In rtic 0.6 using `#[cfg_attr]` ([reference](https://doc.rust-lang.org/reference/conditional-compilation.html#the-cfg_attr-attribute)) seems to allow you to do this.

korken89, 2021-08-20:
Interesting approach, has anyone given this a try?

pdgilbert, 2021-08-20:
I'm in the process of trying it. This works with  release  `cortex-m-rtic = "0.5.3"`:
```
#[cfg_attr(feature = "stm32f1xx", app(device = stm32f1xx_hal::pac, peripherals = true, monotonic = rtic::cyccnt::CYCCNT))]
#[cfg_attr(feature = "stm32f3xx", app(device = stm32f3xx_hal::pac, peripherals = true, monotonic = rtic::cyccnt::CYCCNT))]
```
but I'm not sure I understand the syntax for "the part that you'd normally have to put in `extern "C"` "   and switching from release `rtic` to the git version to get version 0.6 has some other changes I have not yet worked through:
```
use rtic::cyccnt::U32Ext;
             ^^^^^^ could not find `cyccnt` in `rtic`
```
So far I am just trying to compile, no hardware testing yet.

datdenkikniet, 2021-08-20:
In rtic 0.6 (this is using the `master` branch from the github repo), this compiled for me, both with the `f4` and `l4` flags:

```rust
#![no_main]
#![no_std]

use panic_rtt_target as _;

#[cfg_attr(feature = "l4", rtic::app(device = stm32l4xx_hal::pac, dispatchers = [USART1, USART2]))]
#[cfg_attr(feature = "f4", rtic::app(device = stm32f4xx_hal::pac, dispatchers = [DFSDM1, DFSDM2]))]
mod app {
    use dwt_systick_monotonic::DwtSystick;

    #[cfg(feature = "f4")]
    use stm32f4xx_hal::prelude::*;

    #[cfg(feature = "l4")]
    use stm32l4xx_hal::prelude::*;

    #[monotonic(binds = SysTick, default = true)]
    type DwtMono = DwtSystick<80_000_000>;

    #[init]
    fn init(cx: init::Context) -> (init::LateResources, init::Monotonics) {
        let mut rcc = cx.device.RCC.constrain();

        #[cfg(feature = "l4")]
        let mut flash = cx.device.FLASH.constrain();
        #[cfg(feature = "l4")]
        let mut pwr = cx.device.PWR.constrain(&mut rcc.apb1r1);

        let mut dcb = cx.core.DCB;
        let dwt = cx.core.DWT;
        let systick = cx.core.SYST;
        //
        // Initialize the clocks
        //
        #[cfg(feature = "l4")]
        let clocks = rcc.cfgr.sysclk(80.mhz()).freeze(&mut flash.acr, &mut pwr);
        #[cfg(feature = "f4")]
        let clocks = rcc.cfgr.sysclk(80.mhz()).freeze();

        // Setup the monotonic timer
        let mono2 = DwtSystick::new(&mut dcb, dwt, systick, clocks.sysclk().0);
        (init::LateResources {}, init::Monotonics(mono2))
    }

    #[idle]
    fn idle(_: idle::Context) -> ! {
        loop {
            cortex_m::asm::nop();
        }
    }
}
```

pdgilbert, 2021-08-22:
Sorry, I'm struggling with this. I'm a newbie so it is probably something simple. Trying to compile   @datdenkikniet 's code above with 0.6 from master `cortex-m-rtic` `#22ec841ee14cc313b1725ff6ab6de1b5706d2824` I get

```
error: the `#[init]` function must have signature `fn(init::Context) -> (Shared resources struct, Local resources struct, init::Monotonics)`
  --> examples/zzrtic.rs:21:8
   |
21 |     fn init(cx: init::Context) -> (init::LateResources, init::Monotonics) {
   |        ^^^^
```
The only change I've made is to switch "f4" to" stm32f4xx" so it corresponds to my `Cargo.toml`, and to add `dwt-systick-monotonic` to my `Cargo.toml`.

Trying to compile my own example with `cortex-m-rtic` 0.5.3 and conditional compiling of the  `extern "C"` section ( `#[cfg(feature =... ` ) the condition seems to be ignored so interrupts for all MCUs are included and I get

```
error: this extern interrupt is listed more than once
   --> examples/rtic/blink_rtic.rs:401:12
    |
401 |         fn TIM2();
    |            ^^^^

```
So I think 'assuming that using #[cfg(feature = "<l4 or f4>"]  works' is a bad assumption.
I have not been able to figure out proper syntax if `#[cfg_attr(` is suppose to be possible for the `extern "C"` sections.

Also, is there an obvious way to figure out the correspondence between interrupt specifications and `dispatchers`?

datdenkikniet, 2021-08-22:
Ah, seems like the init signature has changed a bit since (or I compiled on an old version). 

An updated version is
<details>
<summary>New rtic 0.6 code</summary>

```rust
#![no_main]
#![no_std]

use panic_rtt_target as _;

#[cfg_attr(feature = "l4", rtic::app(device = stm32l4xx_hal::pac, dispatchers = [USART1, USART2]))]
#[cfg_attr(feature = "f4", rtic::app(device = stm32f4xx_hal::pac, dispatchers = [DFSDM1, DFSDM2]))]
mod app {
    use dwt_systick_monotonic::DwtSystick;

    #[cfg(feature = "f4")]
    use stm32f4xx_hal::prelude::*;

    #[cfg(feature = "l4")]
    use stm32l4xx_hal::prelude::*;

    #[monotonic(binds = SysTick, default = true)]
    type DwtMono = DwtSystick<80_000_000>;

    #[shared]
    struct Shared {}

    #[local]
    struct Local {}

    #[init]
    fn init(cx: init::Context) -> (Shared, Local, init::Monotonics) {
        let mut rcc = cx.device.RCC.constrain();

        #[cfg(feature = "l4")]
        let mut flash = cx.device.FLASH.constrain();
        #[cfg(feature = "l4")]
        let mut pwr = cx.device.PWR.constrain(&mut rcc.apb1r1);

        let mut dcb = cx.core.DCB;
        let dwt = cx.core.DWT;
        let systick = cx.core.SYST;
        //
        // Initialize the clocks
        //
        #[cfg(feature = "l4")]
        let clocks = rcc.cfgr.sysclk(80.mhz()).freeze(&mut flash.acr, &mut pwr);
        #[cfg(feature = "f4")]
        let clocks = rcc.cfgr.sysclk(80.mhz()).freeze();

        // Setup the monotonic timer
        let mono2 = DwtSystick::new(&mut dcb, dwt, systick, clocks.sysclk().0);
        (Shared {}, Local {}, init::Monotonics(mono2))
    }

    #[idle]
    fn idle(_: idle::Context) -> ! {
        loop {
            cortex_m::asm::nop();
        }
    }
}
```

</details>

And indeed, my assumption was wrong. I guess the only way around it would be to find interrupts that are shared and free on all of the platforms that you wish to compile for, or switch to/wait for 0.6 to be released.

The correspondence between interrupt specifications and `dispatchers` you can find in `<pac_crate>::interrupt`, AFAIK, for example [this one for stm32l4](https://docs.rs/stm32l4/0.13.0/stm32l4/stm32l4x1/enum.Interrupt.html).

pdgilbert, 2021-08-22:
Great thanks @datdenkikniet. Your example compiles now, only a linker problem:

```
 = note: rust-lld: error: undefined symbol: _SEGGER_RTT
          >>> referenced by rtt_target.knqqofjh-cgu.4
          >>>               rtt_target-c97207e13e6aa93f.rtt_target.knqqofjh-cgu.4.rcgu.o:(.text._ZN10rtt_target9UpChannel7conjure17h2e8fcf5754da619eE+0x0) in archive /home/paul/githubClones/rust-integration-testing/dev-testing/target/thumbv7em-none-eabi/release/deps/librtt_target-c97207e13e6aa93f.rlib
          >>> referenced by rtt_target.knqqofjh-cgu.4
          >>>               rtt_target-c97207e13e6aa93f.rtt_target.knqqofjh-cgu.4.rcgu.o:(rtt_target::UpChannel::conjure::h2e8fcf5754da619e) in archive /home/paul/githubClones/rust-integration-testing/dev-testing/target/thumbv7em-none-eabi/release/deps/librtt_target-c97207e13e6aa93f.rlib
```
I'll probably stick with 0.6 if I can get to hardware testing.

datdenkikniet, 2021-08-22:
Seems like rust-lld can't find some symbols required for `rtt`. I'm not too familiar with that specific error message, but perhaps you're missing the `rtt_target::rtt_init()`/`rtt_init_print()` call that is required?

pdgilbert, 2021-08-22:
Ok. I changed the panic handler in your example and it links:
```
//use panic_rtt_target as _;
#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;
```
Now to figure out other changes in `cortex-m-rtic` 0.6 to get my example working again. Thanks.

pdgilbert, 2021-08-28:
(Using  version = "0.6.0-alpha.5"     cortex-m-rtic#3ef6508 )  

I have now attempted compiling `cortex-m-rtic` examples `schedule` and `resource-user-struct`, modified as below, using `stm32f1xx_hal`, `stm32f3xx_hal`, `stm32f4xx_hal`, `stm32f7xx_hal`, `stm32h7xx_hal`, and `stm32l4xx_hal`. I have also compiled my example `blink_rtic`, further below, for these HALs and successfully tested it running on a bluepill `stm32f103`, blackpills `stm32f401` and `stm32f411`, and a discovery board `stm32f303`.

Example `schedule` builds for all the above. Example `resource-user-struct` builds for all the above except with `stm32f3xx_hal` the build fails with error
```
66 |         rtic::pend(interrupt::USART1);
   |                               ^^^^^^ variant or associated item not found in `stm32f3xx_hal::interrupt`

error[E0599]: no variant or associated item named `USART2` found for enum `stm32f3xx_hal::interrupt` in the current scope
  --> examples/rtic/resource-user-struct.rs:67:31
   |
67 |         rtic::pend(interrupt::USART2);
   |                               ^^^^^^ variant or associated item not found in `stm32f3xx_hal::interrupt`

error[E0599]: no variant or associated item named `USART1` found for enum `stm32f3xx_hal::interrupt` in the current scope
  --> examples/rtic/resource-user-struct.rs:85:20
   |
85 |     #[task(binds = USART1, shared = [shared])]
   |                    ^^^^^^ variant or associated item not found in `stm32f3xx_hal::interrupt`

error[E0599]: no variant or associated item named `USART2` found for enum `stm32f3xx_hal::interrupt` in the current scope
  --> examples/rtic/resource-user-struct.rs:96:20
   |
96 |     #[task(binds = USART2, shared = [shared])]
   |                    ^^^^^^ variant or associated item not found in `stm32f3xx_hal::interrupt`

```
Suggestions on how to fix that would be appreciated.  I also attempted with `stm32l1xx_hal` but that hal seems to need some work.

<details>
<summary>Click to expand example `schedule`</summary>

```
// from  /cortex-m-rtic/examples/schedule.rs

#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

use panic_semihosting as _;

use rtic::app;

// NOTE: does NOT work on QEMU!

// see <pac>::Interrupt for dispatchers,  e.g. stm32l4::stm32l4x1::Interrupt

#[cfg_attr(feature = "stm32f1xx", app(device = stm32f1xx_hal::pac,   dispatchers = [TIM2, TIM3]))]
#[cfg_attr(feature = "stm32f3xx", app(device = stm32f3xx_hal::pac,   dispatchers = [TIM2, TIM3]))]
#[cfg_attr(feature = "stm32f4xx", app(device = stm32f4xx_hal::pac,   dispatchers = [TIM2, TIM3]))]
#[cfg_attr(feature = "stm32f7xx", app(device = stm32f7xx_hal::pac,   dispatchers = [TIM2, TIM3]))]
#[cfg_attr(feature = "stm32h7xx", app(device = stm32h7xx_hal::pac,   dispatchers = [TIM2, TIM3]))]
#[cfg_attr(feature = "stm32l1xx", app(device = stm32l1xx_hal::stm32, dispatchers = [TIM2, TIM3]))]  //fails
#[cfg_attr(feature = "stm32l4xx", app(device = stm32l4xx_hal::pac,   dispatchers = [TIM2, TIM3]))]

//#[rtic::app(device = lm3s6965, dispatchers = [SSI0])]

mod app {
    use cortex_m_semihosting::hprintln;
    use dwt_systick_monotonic::DwtSystick;
    use rtic::time::duration::Seconds;

    const MONO_HZ: u32 = 8_000_000; // 8 MHz

    #[monotonic(binds = SysTick, default = true)]
    type MyMono = DwtSystick<MONO_HZ>;

    #[shared]
[... 511 lines cut ...]
    #[task(shared = [led])]
    fn blink(_cx: blink::Context, duration: Milliseconds<u32>) {
        // note that if blink is called with ::spawn_after then the first agument is the after time
        // and the second is the duration.
        crate::app::led_off::spawn_after(duration).ok();
        crate::app::led_on::spawn().ok();
    }
   

    #[task(shared = [led])]
    fn led_on(mut cx: led_on::Context) {
        cx.shared.led.lock(|led| led.on());
    }
    

    #[task(shared = [led])]
    fn led_off(mut cx: led_off::Context) {
        cx.shared.led.lock(|led| led.off());
    }
}

```
</details>

CI testing of these and other examples  is under tab "Actions"  at https://github.com/pdgilbert/rust-integration-testing/.  (It is best to look at the "dev-testing" versions of results.)

pdgilbert, 2021-09-06:
I have now tested the above discussed `#[cfg_attr(feature = ...,  rtic::app(...`  new feature in `cortex-m-rtic 0.6` with a few examples and device crates and it seems to work well. CI for these is running at https://github.com/pdgilbert/rust-integration-testing/actions. (See dev-testing jobs.) The examples are `blink_rtic`,  `resource-user-struct`,  `schedule`, `ccs811-gas-voc-usart-logger` and `iaq-core-c-gas-voc-usart-logger`. The last two are modified from @eldruin 's examples at https://github.com/eldruin/driver-examples. The examples compile with `stm32f1xx`, `stm32f4xx`, `stm32f7xx`, and `stm32h7x`x. I have tested that the `blink_rtic` example runs on bluepill and blackpills with stm32f401 and stm32f411. There remain problems compiling for `stm32f3xx`, `stm32l1xx` and `stm32l4xx`. I will post a new issue on these when I examine them more carefully.

closed 2021-09-06 by pdgilbert