rtic-rs/rtic #499: CI: consider using xtask

japaric, 2021-07-22:
Currently, the CI test suite is executed by GitHub Actions by reading a shell script that depends on environment variables and is [embedded in a YAML file](https://github.com/rtic-rs/cortex-m-rtic/blob/2f3b5cba805d7e7b736869249f46298e59bc944d/.github/workflows/build.yml#L202-L332). It's not easy to run the same tests (less the full test suite) on a local machine, not even on a *NIX OS.

I suggest repackaging the test suite into [Cargo xtasks](https://github.com/matklad/cargo-xtask) (+). Such that contributors only need to run `cargo xtask test full` locally to run the *complete* test suite for the relevant target matrix (thumbv6m & thumbv7em, I guess). There would be smaller xtasks like `cargo xtask test run-pass` and `cargo xtask test style`, etc. that would be run from GHA jobs. These smaller xtasks should be configurable via command-line flags so they can be invoked like `cargo xtask test run-pass --target=${{ matrix.target }}` from the GHA jobs.

(+) A Cargo xtask is, in a nutshell, "shell scripts and Makefiles but in Rust"

perlindgren, 2021-08-10:
Sounds like a very good idea.

korken89, 2021-08-10:
@japaric do you feel like making an example for it from an existing test to help us on the way? We've never used it before.

japaric, 2021-08-12:
We are using xtask in the defmt repo. I didn't write it so I can't comment much on the details but the xtask source code is [here](https://github.com/knurling-rs/defmt/blob/f5abeacc666ae942a6483b30222b0175f63c97f9/xtask/src/main.rs) and could be used as a reference.

You also need to add [this](https://github.com/knurling-rs/defmt/blob/f5abeacc666ae942a6483b30222b0175f63c97f9/.cargo/config.toml#L1-L2) to `.cargo/config` and add the xtask crate to [the workspace](https://github.com/knurling-rs/defmt/blob/f5abeacc666ae942a6483b30222b0175f63c97f9/Cargo.toml#L46). Once that's in place you can run e.g. `cargo xtask test-cross` to run cross-compiled tests on QEMU (iirc, this also does `rustup target thumbv7-none-eabi` under the hood so it takes care of installing Rust dependencies).

korken89, 2021-08-19:
I see now how this works, thanks for the references!
I thought it was a cargo command, but I see now it's a template.

I'll have a look at converting. Maybe shamelessly stealing your example code :D

closed 2021-09-21 by bors[bot]