rtic-rs/rtic #505: Improve modularity by allowing the `#[init]` task to be externed

theunkn0wn1, 2021-07-27:
In RTIC 0.6, it is possible to extern `#[task]` implementations so they may be implemented in other modules.
Following this logic, I reasoned it should be possible to extern the `#[init]` task as well. Unfortunately attempting this results in compilation errors.

```rs
#![deny(unsafe_code)]
#![no_main]
#![no_std]
#![allow(unused_imports)]

use panic_rtt_target as _panic_handler;

/* declare a submodule for handling tim8 interrupts */
mod tim8;
mod init;

/* declare the RTIC application itself */
#[rtic::app(device = stm32f4xx_hal::stm32, peripherals = true)]
mod app {
    /* bring dependencies into scope */
    use rtt_target::{rprintln, rtt_init_print};
    use stm32f4xx_hal::{
        gpio::{gpioc::PC6, Alternate},
        prelude::*,
        pwm_input::PwmInput,
        stm32::TIM8,
        timer::Timer,
    };

    /// PWM input monitor type
    pub(crate) type PwmMonitor = PwmInput<TIM8, PC6<Alternate<3>>>;

    /* resources shared across RTIC tasks */
    #[shared]
    struct Shared {
        /// the last observed position of the turret
        last_observed_turret_position: f32,
    }

    /* resources local to specific RTIC tasks */
    #[local]
    struct Local {
        monitor: PwmMonitor,
    }

    /* bring tim8's interrupt handler into scope */
    use crate::tim8::tim8_cc;
    use crate::init::init;

    // RTIC docs specify we can modularize the code by using these `extern` blocks.
    // This allows us to specify the interrupt handlers in other modules and still work within
    // RTIC's infrastructure.
    extern "Rust" {
        #[task(binds = TIM8_CC, local = [monitor], shared = [last_observed_turret_position])]
        fn tim8_cc(context: tim8_cc::Context);
        #[init]
        fn init(ctx: init::Context) -> (Shared, Local, init::Monotonics);
    }
}
```

Expected result: compile succeeds and functions as expected

Actual result: compile failure.
```
error: `extern` task required `#[task(..)]` attribute
  --> src/main.rs:53:12
   |
53 |         fn init(ctx: init::Context) -> (Shared, Local, init::Monotonics);
   |            ^^^^
   ```

perlindgren, 2021-08-10:
Yea, this is a known issue, it has not yet been implemented. We don't see any obvious problems of doing so. It might also be added later as a non-breaking change.

AfoHT, 2023-07-05:
Closed by https://github.com/rtic-rs/rtic/pull/767 :tada:

closed 2023-07-05 by AfoHT