rtic-rs/rtic #510: Late resource example in 0.5 is missing unsafe to use static mut Queue

tdaede, 2021-08-15:
On this book page: https://rtic.rs/0.5/book/en/by-example/resources.html

```
    fn init(_: init::Context) -> init::LateResources {
        static mut Q: Queue<u32, U4> = Queue(i::Queue::new());

        let (p, c) = Q.split();

        // Initialization of late resources
        init::LateResources { p, c }
    }
```
I don't see how this can work without unsafe{} around the Q.split() operation.

korken89, 2021-08-16:
Hi,

This uses the same static mut transform of RTIC 0.5.
If the first lives of a task uses static mut it will perform a coffee transform to make it safe.
This has been removed in 0.6 in favour of local resources.

tdaede, 2021-08-16:
Aha, indeed I didn't even realize that transform existed. I tried to replicate the example but put the static mut declaration in a place where it wasn't transformed.

closed 2021-08-16 by tdaede