rtic-rs/rtic #517: trait implementation fails on task shared object

pdgilbert, 2021-08-25:
(Using 0.6-alpha from current github #22ec841.) I am trying to implement a trait but the implementation is not recognized for the shared object. The following code compiles with the `cx.shared.led.on()` line commented out
<details>
<summary>Click to expand</summary>

```
#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use rtic::app;

#[cfg_attr(feature = "stm32l4xx", app(device = stm32l4xx_hal::pac,   peripherals = true, dispatchers = [USART1, USART2]))]

mod app {
    use dwt_systick_monotonic::DwtSystick;
    use rtic::time::duration::{Seconds, Milliseconds};

    const PULSE: u32 = 1; // 1 second
    const PERIOD: u32 = 10 * PULSE; // 10 seconds
    
    use stm32l4xx_hal::{
        gpio::{gpioc::PC13, Output, PushPull},
        pac::Peripherals,
        prelude::*,
    };

    use embedded_hal::digital::v2::OutputPin;
    
    type LedType = PC13<Output<PushPull>>;
    
    fn setup(dp: Peripherals) -> LedType {
        let mut rcc   = dp.RCC.constrain();
        let mut gpioc = dp.GPIOC.split(&mut rcc.ahb2);
        let led = gpioc.pc13.into_push_pull_output(&mut gpioc.moder, &mut gpioc.otyper);

        impl LED for PC13<Output<PushPull>> {
            fn on(&mut self)  -> () { self.set_low().unwrap() }
            fn off(&mut self) -> () { self.set_high().unwrap()}
        }

        led
    }
        
    pub trait LED {
        fn on(&mut self) -> ();
        fn off(&mut self) -> ();
    }
    
    #[monotonic(binds = SysTick, default = true)]
    type DwtMono = DwtSystick<80_000_000>;

    #[local]
    struct Local {}
    
    #[init]
    fn init(cx: init::Context) -> (Shared, Local, init::Monotonics) {
        let mut dcb = cx.core.DCB;
        let dwt = cx.core.DWT;
        let systick = cx.core.SYST;
    
        let mono = DwtSystick::new(&mut dcb, dwt, systick, 8_000_000);

        let device = cx.device;
    
        let mut led = setup(device);
    
        led.off();   //WORKS HERE
        led.on();
        led.off();

        crate::app::one::spawn_after(Milliseconds(PULSE * 1000 - 700)).ok();
    
       (Shared {led}, Local {}, init::Monotonics(mono))
    }
    
    #[shared]
    struct Shared {
        led: LedType,
    }
    
    #[task(shared = [led])]
    fn one(cx: one::Context) {
        //cx.shared.led.on();   //FAILS HERE  compiles with this line commented out
        crate::app::one::spawn_after(Seconds(PULSE)).ok();
    }

 }
```
</details>

but with the line uncommented I get
```
3 | #[cfg_attr(feature = "stm32l4xx", app(device = stm32l4xx_hal::pac,   peripherals = true, dispatchers = [USART1, USART2]))]
   |                                   -------------------------------------------------------------------------------------- method `on` not found for this
...
85 |         cx.shared.led.on();   //FAILS HERE  compiles with this line commented out
   |                       ^^ method not found in `led<'_>`
   |
   = help: items from traits can only be used if the trait is implemented and in scope
note: `LED` defines an item `on`, perhaps you need to implement it
  --> examples/zzled.rs:46:5
   |
46 |     pub trait LED {
   |     ^^^^^^^^^^^^^
```
The implementation works in the `#[init]` section, it is only in the `#[task(shared = [led])]` section that I get the error.

I've been testing mostly with `stm32l4xx` but it also occurs with `stm32f1xx`.

korken89, 2021-08-25:
Hi,

You need to lock the led fits.

pdgilbert, 2021-08-28:
A more complete working and run tested example, from which the above was extracted, is now at #490.