rtic-rs/rtic #521: Unused `dispatchers` are not checked for validity

starthal, 2021-08-30:
The `task` example requires two dispatchers.

When a required dispatcher is invalid, e.g. 
```rust
#[rtic::app(device = lm3s6965, dispatchers = [SSI0, FOO])]
```
behavior is expected:
```
error[E0599]: no variant or associated item named `FOO` found for enum `Interrupt` in the current scope
  --> examples\task.rs:10:53
   |
10 | #[rtic::app(device = lm3s6965, dispatchers = [SSI0, FOO])]
   |                                                     ^^^ variant or associated item not found in `Interrupt`
```

However, when the dispatcher is not needed (yet):
```rust
#[rtic::app(device = lm3s6965, dispatchers = [SSI0, QEI0, FOO])]
```

The build succeeds, unexpectedly. When another priority is used, this will cause an unexpected error.

AfoHT, 2021-08-31:
I can see how this can cause some confusion when a _unrelated_ error pops up when changing something, but I'm not sure it is really a big problem, considering when it does give you an error it is pointing out the offending dispatcher nicely.

I find it fairly impressive that when adding something unused it will be completely pruned and the compiler will not have to worry at all.

Not sure if there is a smart way to get earlier checking than what is currently available or if it is worth the effort.

Thoughts? @korken89 @perlindgren

perlindgren, 2021-08-31:
I'm not sure in this case, but in general one should strive to reject errors as early as possible, such to prevent unrelated changes to raise the error. (Like adding a soft task with unique priority).

AfoHT, 2021-08-31:
Do you know if we have access to the HAL interrupt vectors during the macro-stage of compilation?

perlindgren, 2021-08-31:
I suspect the error emerges later, so I'm not sure we have the info to fix it earlier.

commit 1a49d674 (2021-08-31): validate unused dispatchers

closes #521
commit f46b22f7 (2021-08-31): Merge #522

522: validate unused dispatchers r=AfoHT a=japaric

closes #521
PR #522 (2021-08-31): validate unused dispatchers
closes #521
closed 2021-08-31 by bors[bot]