rtic-rs/rtic #523: SYST ownership

pdgilbert, 2021-09-02:
(Using `cortex-m-rtic`#f46b22f  version 0.6. )   The example code below builds but I cannot understand why there is no ownership conflict between `SYST` used for `delay` in the `setup()` and used in `DwtSystick`.

<details>
<summary>Click to expand</summary>

```
//  Why is there not a conflict for SYST between delay and DwtSystick

#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use rtic::app;

#[cfg_attr(feature = "stm32f1xx", app(device = stm32f1xx_hal::pac,   dispatchers = [TIM2, TIM3]))]

mod app {

    use dwt_systick_monotonic::DwtSystick;
    use rtic::time::duration::{Seconds, };
 
    const PERIOD: Seconds = Seconds(10);
   

    #[cfg(feature = "stm32f1xx")]
    use stm32f1xx_hal::{
        delay::Delay,
        gpio::{ gpioc::PC13, Output, PushPull, },
        pac::Peripherals,
        prelude::*,
    };
    
    #[cfg(feature = "stm32f1xx")]
    const CLOCK: u32 = 8_000_000; //should be set for board not for HAL

    #[cfg(feature = "stm32f1xx")]
    type LedType = PC13<Output<PushPull>>;
    
//    use cortex_m::Peripherals as CorePeripherals;
    
    #[cfg(feature = "stm32f1xx")]
    fn setup(dp: Peripherals) -> (LedType, Delay) {
//    fn setup(dp: Peripherals, cp: &CorePeripherals) -> (LedType, Delay) {

        let mut flash = dp.FLASH.constrain();
        let rcc = dp.RCC.constrain();
        let clocks = rcc
            .cfgr
            .use_hse(8.mhz())
            .sysclk(72.mhz())
            .pclk1(36.mhz())
            .freeze(&mut flash.acr);
    
        // Why does next not cause buld problem with  DwtSystick::new  in init?
// comment out next when cp passed in argument
        let cp = cortex_m::Peripherals::take().unwrap();
        let delay = Delay::new(cp.SYST, clocks);
    
   
        let mut gpioc = dp.GPIOC.split();
        let mut led = gpioc.pc13.into_push_pull_output(&mut gpioc.crh);
 
        impl LED for PC13<Output<PushPull>> {
            fn on(&mut self) -> () {
                self.set_low()
            }
            fn off(&mut self) -> () {
                self.set_high()
            }
        }
        led.off();
    
        (led, delay)
    }
    
       
    pub trait LED {
        fn on(&mut self) -> ();
        fn off(&mut self) -> ();
    }
    
    #[monotonic(binds = SysTick, default = true)]
    type DwtMono = DwtSystick<CLOCK>;        
       
    #[shared]
    struct Shared { led: LedType }
    
    #[local]
    struct Local { led_state: bool }

    #[init]
    fn init(mut cx: init::Context) -> (Shared, Local, init::Monotonics) {

        let (mut led, mut delay) = setup(cx.device);
//        let (mut led, mut delay) = setup(cx.device, &cx.core);

        let mono = DwtSystick::new(&mut cx.core.DCB, cx.core.DWT, cx.core.SYST, CLOCK);

        led.off();
        let led_state: bool = false;
        
        // in real example this is a delay while something initializes
        delay.delay_ms(10_u16);

        toggle::spawn_after(PERIOD).unwrap();

        (Shared {led: led, }, Local {led_state: led_state, },  init::Monotonics(mono))
     }

    #[task(shared = [led], local = [led_state])]
    fn toggle(mut cx: toggle::Context) {

        if *cx.local.led_state {
            cx.shared.led.lock(|led| led.off());
            *cx.local.led_state = false;
        } else {
            cx.shared.led.lock(|led| led.on());
            *cx.local.led_state = true;
        }

        toggle::spawn_after(PERIOD).unwrap();
    }

}
```
</details>

If I change the few commented lines to pass `cx.core` to `setup()`, the way I think it should need to be done, then I get an error:

```
cannot move out of `cp.SYST` which is behind a shared reference
      let delay = Delay::new(cp.SYST, clocks);
                                   ^^^^^^^ move occurs because `cp.SYST` has type `SYST`, which does not implement the `Copy` trait
```

I have also tried variations on this but not found the way to make it work.  I would appreciate it if someone can explain why the way that compiles actual should be good, or an indication of the correct way to do it. TIA.

korken89, 2021-09-03:
Hi, this will compile but it will `panic` on the line

```
let cp = cortex_m::Peripherals::take().unwrap();
```

In the start of RTIC we do this already, so the `cortex_m::Peripherals` is marked as taken - which means `take()` will return `None` and then panic.

The main issue here is that you want to use SysTick as an RTIC Monotonic and `Delay` at the same time, this will not work as both takes ownership of the SysTick peripheral.
Either you need to use a `Delay` implementation that is not based on SysTick or an RTIC Monotonic that is not based on SysTick.

pdgilbert, 2021-09-03:
Ok, that is the what I suspected to be a problem. It would be nice if this could be caught by the  compiler rather than throw a run time panic.

closed 2021-09-03 by pdgilbert

pdgilbert, 2021-09-03:
(Newbie question.)  I just want the delay while a sensor initializes, before I spawn anything. Is there a simple way to use RTIC Monotonic for a delay in `init` before I start other things?

mattico, 2021-09-03:
I would either use Delay (https://docs.rs/embedded-hal/0.2.6/embedded_hal/blocking/delay/index.html) with a timer, or use `cortex_m::asm::delay`.

pdgilbert, 2021-09-06:
Thanks @korken89 and @mattico. Code from above modified as below using `cortex_m::asm::delay` compiles and runs on bluepill. (I have not figured out how to use `emmbedded_hall::blocking::delay` with anything but the default systick clock. Pointers on that would be appreciated.) I also verified that code above does panic.

Using `cortex-m-rtic` version 0.6 features discussed in #490 I have a more complete example of this called `blink_rtic`, which is compiling with `stm32f1xx`, `stm32f4xx`, `stm32f7xx`, and `stm32h7xx`. I have tested that it runs on bluepill and blackpills with stm32f401 and stm32f411. This example is maintained with other and CI at https://github.com/pdgilbert/rust-integration-testing/actions.

The example in the original post compiles and runs on bluepill when modified as follows:
<details>
<summary>Click to expand</summary>

```
#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use rtic::app;

#[cfg_attr(feature = "stm32f1xx", app(device = stm32f1xx_hal::pac,   dispatchers = [TIM2, TIM3]))]

mod app {

    use dwt_systick_monotonic::DwtSystick;
    use rtic::time::duration::{Seconds, };

    use cortex_m::asm; //asm::delay(N:u32) blocks the program for at least N CPU cycles.
    //delay_ms could be used but needs to use a timer other than Systick
    //use embedded_hal::blocking::delay; //delay::delay_ms(N:u32) blocks the program for N ms.

    const PERIOD: Seconds = Seconds(10);
   

    #[cfg(feature = "stm32f1xx")]
    use stm32f1xx_hal::{
        gpio::{ gpioc::PC13, Output, PushPull, },
        pac::Peripherals,
        prelude::*,
    };
    
    #[cfg(feature = "stm32f1xx")]
    const CLOCK: u32 = 8_000_000; //should be set for board not for HAL

    #[cfg(feature = "stm32f1xx")]
    type LedType = PC13<Output<PushPull>>;
        
    #[cfg(feature = "stm32f1xx")]
    fn setup(dp: Peripherals) -> LedType {

        let mut gpioc = dp.GPIOC.split();
        let mut led = gpioc.pc13.into_push_pull_output(&mut gpioc.crh);
 
        impl LED for PC13<Output<PushPull>> {
            fn on(&mut self) -> () {
                self.set_low()
            }
            fn off(&mut self) -> () {
                self.set_high()
            }
        }
        led.off();
    
        led
    }
    
       
    pub trait LED {
        fn on(&mut self) -> ();
        fn off(&mut self) -> ();
    }
    
    #[monotonic(binds = SysTick, default = true)]
    type DwtMono = DwtSystick<CLOCK>;        
       
    #[shared]
    struct Shared { led: LedType }
    
    #[local]
    struct Local { led_state: bool }

    #[init]
    fn init(mut cx: init::Context) -> (Shared, Local, init::Monotonics) {

        let mut led = setup(cx.device);
        
        // In an application this is a delay while something initializes.
        // Note that this delay cannot use SYST because Monotonics uses that (for spawn,
        //   although spawn has not yet happened so there may be a way?)
        //   delay_ms() needs to use a timer other than default Systick

        led.on();
        asm::delay(2 * CLOCK); // (2 * CLOCK cycles give aprox 2 second delay
        //delay::delay_ms(2_000_u16);
        led.off();
        let led_state: bool = false;
        
        let mono = DwtSystick::new(&mut cx.core.DCB, cx.core.DWT, cx.core.SYST, CLOCK);

        toggle::spawn_after(PERIOD).unwrap();

        (Shared {led: led, }, Local {led_state: led_state, },  init::Monotonics(mono))
     }

    #[task(shared = [led], local = [led_state])]
    fn toggle(mut cx: toggle::Context) {

        if *cx.local.led_state {
            cx.shared.led.lock(|led| led.off());
            *cx.local.led_state = false;
        } else {
            cx.shared.led.lock(|led| led.on());
            *cx.local.led_state = true;
        }

        toggle::spawn_after(PERIOD).unwrap();
    }

}

```
</details>

prutschman, 2022-06-15:
Apologies for the necro-post, but in case someone comes across this thread via google like I did, I found the following way to "re-use" the DwtSystick for general initialization delays before it gets used by the RTIC runtime. I wanted something that implemented `embedded_hal::blocking::delay::DelayMs<_>` so I could have my initialization functions take an `impl DelayMs`, but pass it something based on DwtSystick.

```
    type MyMono = DwtSystick<MONO_HZ>;

    struct DelayWrapper<'a>(&'a mut MyMono);

    impl embedded_hal::blocking::delay::DelayMs<u16> for DelayWrapper<'_> {
        fn delay_ms(&mut self, ms: u16) {
            let wait_until = self.0.now() + (ms as u32).millis();
            while self.0.now() < wait_until {
                /* spin */
            }
        }
    }
    ```
    
    Then later,
    
    ```
codec.initialize(&mut DelayWrapper(&mut mono));
```

But you can still pass `mono` into the runtime afterward.