rtic-rs/rtic #524: Compiler unable to determine Lifetime of local static mut in 0.6.0-alpha.5

LongLiveCHIEF, 2021-09-04:
I'm upgrading from `0.6.0-alpha.2` to `0.6.0-alpha.5`.  The last bit i'm struggling with is one of my former `static mut` resources, which is now a `local` resource.

Whenever I try to use the technique suggested in the migration docs (see: [no static mut transform](https://rtic.rs/dev/book/en/migration/migration_v5.html#no-static-mut-transform)), I get a compiler error that says `lifetime 'static required'

taken from: https://github.com/LongLiveCHIEF/led-backlighting-controller/blob/b718a3af2c510bfc8706e211e1cc812a95a50a1a/src/main.rs#L123-L148

```rs
    #[task(shared = [modeDetectPin], local = [
        hold: Option<hold::SpawnHandle> = None,
        pressed_at: Option<Instant<RtcMonotonic>> = None
    ])]
    fn debounce(mut cx: debounce::Context){
        let HOLD: &'static mut Option<hold::SpawnHandle> = cx.local.hold;
        let PRESSED_AT: &'static mut Option<Instant<RtcMonotonic>> = cx.local.pressed_at;
        if let Some(handle) = HOLD.take() {
            handle.cancel().ok();
        }
```

Compiling results in the following error:

```rs
error[E0621]: explicit lifetime required in the type of `cx`
   --> src/main.rs:128:60
    |
128 |         let HOLD: &'static mut Option<hold::SpawnHandle> = cx.local.hold;
    |                                                            ^^^^^^^^^^^^^ lifetime `'static` required

error: aborting due to previous error
```
If I try to specify the lifetime of `cx: debounce::Context`, i get a compiler `error: unexpected lifetime 'static in pattern`.

Rust version:  `rustc 1.54.0 (a178d0322 2021-07-26)`

LongLiveCHIEF, 2021-09-04:
Getting rid of type and lifetime annotations for the `HOLD` and `PRESSED_AT` vars allows the project to compile, and it works as expected. Just as an FYI.  It's not a solution I'm happy with, but one that'll work in the short term.

LongLiveCHIEF, 2021-09-09:
I missed the weekly meetup this week, but saw you guys asked if I was around.  For my part, my only real concern about this is that the docs aren't discontiguous around the need to explicitly specify lifetime for these, so that we're friendlier to beginners to Rust.

I was just certain I was doing something wrong and just removed the annotations for lifetimes as a last ditch effort to see if the compiler could figure out what it was I wanted to do.

korken89, 2021-09-22:
Hi, as we discussed this is expected behavior - there is no need to specify lifetimes when using locals.
If you have any other issue, feel free to ping!

closed 2021-09-22 by korken89