rtic-rs/rtic #527: refactor: move test metadata into test file source code

japaric, 2021-09-16:
something that came up during PR #526 is that there's test specific information hard coded in the xtask logic.
this information could be moved into the test file source code (`examples/*.rs`)

the Rust compiler test suite uses this approach: it encodes test metadata in code comments.

for the particular case of RTIC the information that we need to encode is (off the top of my head)
- whether the test is ARMv7 only
- whether the example should NOT be run in QEMU (no CYCCNT in QEMU = spawn_at API doesn't work)

what this could look like is (straw man proposal)

``` console
$ head -n1 examples/t-schedule.rs
// skip-run-test
```

that's the metadata for compile this example but don't run it in QEMU

``` console
$ head -n1 examples/pool.rs
// v7-only
```

that's the metadata for "only compile this for the thumbv7m-none-eabi target"

an advantage of this approach is that you don't need to list the examples in the `xtask` source code, which is bound to get out of date
instead you can have the `xtask` scan *all*  the files in `examples/` and decide what to do with the example (compile-pass for v6 and v7; run-pass; compile-pass for v7 only; etc.) based on the metadata

AfoHT, 2021-09-21:
This is quite nice approach, as updating the xtasks for each added/removed example isn't ideal.

Something to keep in mind is that all examples are included into the book, so anything added there will be in the book too.

AfoHT, 2021-09-21:
#528 Removes the need for special handling of CYCCNT, allows running all examples in QEMU :+1:

japaric, 2021-09-23:
> Something to keep in mind is that all examples are included into the book, so anything added there will be in the book too.

mdbook lets you [include portions of a file](https://rust-lang.github.io/mdBook/format/mdbook.html#including-portions-of-a-file) so we could put the metadata comment at the top and not include that first line in the book.

commit f0c31998 (2021-09-25): Merge #528

528: The great 0.6 docs update r=AfoHT a=korken89

Closes #530
Closes #527
Closes #487
Closes #461
Closes #448
Closes #440
Closes #422
PR #528 (2021-09-25): The great 0.6 docs update
Closes #530
Closes #527
Closes #487
Closes #461
Closes #448
Closes #440
Closes #422
closed 2021-09-28 by AfoHT