rtic-rs/rtic #540: stm32f3xx_hal  interrupt::USART

pdgilbert, 2021-10-01:
I have a modified version of the `resources-user-struct` example building with `stm32f1xx_hal`, `stm32f4xx_hal`, `stm32f7xx_hal`, `stm32h7xx_hal` and `stml4xx_hal`

<details>
<summary>Click to expand</summary>

```
//  from  /cortex-m-rtic/examples/resource.rs

#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

use panic_semihosting as _;

use rtic::app;

#[cfg_attr(feature = "stm32f1xx", app(device = stm32f1xx_hal::pac  ))]
#[cfg_attr(feature = "stm32f3xx", app(device = stm32f3xx_hal::pac  ))]  //fails:  USART1 variant or associated item not found in `stm32f3xx_hal::interrupt
#[cfg_attr(feature = "stm32f4xx", app(device = stm32f4xx_hal::pac  ))]
#[cfg_attr(feature = "stm32f7xx", app(device = stm32f7xx_hal::pac  ))]
#[cfg_attr(feature = "stm32h7xx", app(device = stm32h7xx_hal::pac  ))]
#[cfg_attr(feature = "stm32l1xx", app(device = stm32l1xx_hal::stm32))]  //fails: trait `core::marker::Copy` is not implemented for `Interrupt
#[cfg_attr(feature = "stm32l4xx", app(device = stm32l4xx_hal::pac  ))]

//#[rtic::app(device = lm3s6965)]

mod app {
    use cortex_m_semihosting::{debug, hprintln};

    //use lm3s6965::Interrupt;

    #[cfg(feature = "stm32f1xx")]
    use stm32f1xx_hal::pac::Interrupt  as interrupt;

    #[cfg(feature = "stm32f3xx")]
    use stm32f3xx_hal::interrupt;

    #[cfg(feature = "stm32f4xx")]
    use stm32f4xx_hal::interrupt;

    #[cfg(feature = "stm32f7xx")]
    use stm32f7xx_hal::interrupt;

    #[cfg(feature = "stm32h7xx")]
    use stm32h7xx_hal::interrupt;

    #[cfg(feature = "stm32l1xx")]
    use stm32l1xx_hal::interrupt;

    #[cfg(feature = "stm32l4xx")]
    use stm32l4xx_hal::interrupt;

    #[shared]
    struct Shared {
        // A resource
        shared: u32,
    }

    // Should not collide with the struct above
    #[allow(dead_code)]
    struct Shared2 {
        // A resource
        shared: u32,
    }

    #[local]
    struct Local {}

    #[init]
    fn init(_: init::Context) -> (Shared, Local, init::Monotonics) {
        rtic::pend(interrupt::USART1);
        rtic::pend(interrupt::USART2);

        (Shared { shared: 0 }, Local {}, init::Monotonics())
    }

    // `shared` cannot be accessed from this context
    #[idle]
    fn idle(_cx: idle::Context) -> ! {
        hprintln!("about to  debug::EXIT_SUCCESS").unwrap();
        debug::exit(debug::EXIT_SUCCESS);

        // error: no `shared` field in `idle::Context`
        // _cx.shared.shared += 1;

        loop {}
    }

    // `shared` can be accessed from this context
    #[task(binds = USART1, shared = [shared])]
    fn usart1(mut cx: usart1::Context) {
        let shared = cx.shared.shared.lock(|shared| {
            *shared += 1;
            *shared
        });

        hprintln!("USART1: shared = {}", shared).unwrap();
    }

    // `shared` can be accessed from this context
    #[task(binds = USART2, shared = [shared])]
    fn usart2(mut cx: usart2::Context) {
        let shared = cx.shared.shared.lock(|shared| {
            *shared += 1;
            *shared
        });

        hprintln!("USART2: shared = {}", shared).unwrap();
    }
}

```
</details>

but on `stm32f3xx_hal` I get
```
   Compiling rust-integration-testing-of-examples v0.1.0 (/home/paul/githubClones/rust-integration-testing/dev-testing)
error[E0599]: no variant or associated item named `USART1` found for enum `stm32f3xx_hal::interrupt` in the current scope
  --> examples/rtic/resource-user-struct.rs:66:31
   |
66 |         rtic::pend(interrupt::USART1);
   |                               ^^^^^^ variant or associated item not found in `stm32f3xx_hal::interrupt`

error[E0599]: no variant or associated item named `USART2` found for enum `stm32f3xx_hal::interrupt` in the current scope
  --> examples/rtic/resource-user-struct.rs:67:31
   |
67 |         rtic::pend(interrupt::USART2);
   |                               ^^^^^^ variant or associated item not found in `stm32f3xx_hal::interrupt`

error[E0599]: no variant or associated item named `USART1` found for enum `stm32f3xx_hal::interrupt` in the current scope
  --> examples/rtic/resource-user-struct.rs:85:20
   |
85 |     #[task(binds = USART1, shared = [shared])]
   |                    ^^^^^^ variant or associated item not found in `stm32f3xx_hal::interrupt`

error[E0599]: no variant or associated item named `USART2` found for enum `stm32f3xx_hal::interrupt` in the current scope
  --> examples/rtic/resource-user-struct.rs:96:20
   |
96 |     #[task(binds = USART2, shared = [shared])]
   |                    ^^^^^^ variant or associated item not found in `stm32f3xx_hal::interrupt`

error: aborting due to 4 previous errors

For more information about this error, try `rustc --explain E0599`.
error: could not compile `rust-integration-testing-of-examples`
```

I've looked in a few obvious place but cannot find the usart interrupts. Any suggestions?

mattico, 2021-10-02:
Looks like the interrupts are shared with some EXTI channels: https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/enum.interrupt.html#variant.USART1_EXTI25

pdgilbert, 2021-10-03:
I can make this work with `stm32f3xx_hal` by using `USART1_EXTI25` everywhere. Unfortunately that breaks all the other hals. The
```
        rtic::pend(interrupt::USART1);
        rtic::pend(interrupt::USART2);
```
can be handled with
```
    #[cfg(feature = "stm32f3xx")]
    mod interrupt {
      pub use stm32f3xx_hal::interrupt::*;
      pub use stm32f3xx_hal::interrupt::{USART1_EXTI25 as USART1, USART2_EXTI26  as USART2}; 
    }
```

Unfortunately, I do not see any way to handle
```
    #[task(binds = USART1, shared = [shared])]
    fn usart1(mut cx: usart1::Context) {
   ...
```
Since  `binds = USART1` seems to be looking directly at `stm32f3xx_hal` I still get
```Compiling rust-integration-testing-of-examples v0.1.0 (/home/paul/githubClones/rust-integration-testing/dev-testing)
error[E0599]: no variant or associated item named `USART1` found for enum `stm32f3xx_hal::interrupt` in the current scope
  --> examples/rtic/resource-user-struct.rs:95:20
   |
95 |     #[task(binds = USART1, shared = [shared])]
   |                    ^^^^^^ variant or associated item not found in `stm32f3xx_hal::interrupt`

error[E0599]: no variant or associated item named `USART2` found for enum `stm32f3xx_hal::interrupt` in the current scope
   --> examples/rtic/resource-user-struct.rs:106:20
    |
106 |     #[task(binds = USART2, shared = [shared])]
    |                    ^^^^^^ variant or associated item not found in `stm32f3xx_hal::interrupt`

error: aborting due to 2 previous errors
```

Any suggestions?

korken89, 2021-10-04:
AFAICT this is not supported.
Unfortunately with the `#[cfg]` expansion support being as bad as it is in proc-macros, I'd not expect this to be solved anytime soon. 

What you probably can do though, is to make a trampoline function with some `unsafe C` - but I'm not sure if this is preferable in your case.
Something like this.

```rust
#[no_mangle]
#[inline(never)]
unsafe extern "C" fn USART1() {
    USART1_EXTI25();
}
```

perlindgren, 2021-11-30:
We close this issue for now, see [#197](https://github.com/rtic-rs/cortex-m-rtic/issues/197)

closed 2021-11-30 by perlindgren