rtic-rs/rtic #541: Or-pattern support

antonok-edm, 2021-10-05:
The following function definition uses an "or pattern" and compiles as of [Rust 1.53.0](https://blog.rust-lang.org/2021/06/17/Rust-1.53.0.html#or-patterns).

```rust
fn or_pattern_issue() {
    struct A;

    enum B {
        B1,
        B2,
    }

    match (A, B::B1) {
        (A, B::B1 | B::B2) => (),
    }
}
```

However, placing the exact same function definition (plus a `#[task(schedule = [], resources = [])]` attribute and a `_: or_pattern_issue::Context` argument) inside of an `#[rtic::app(...)]` block fails to compile. The error explains that the line inside of the `match` block expects a `,` at the location of the `|` character.

Manually expanding the `match` arm into two separate ones works, but is tedious in a simple case like this and quickly becomes unmaintainable with more significant logic.

I've tried this on v5.9. I haven't verified whether or not it applies to v6.0, but it would be great to have or-patterns supported on the stable branch.

perlindgren, 2021-10-05:
Hi. Interesting. Have you an example of the RTIC code and we can take a look.

/Per

antonok-edm, 2021-10-05:
Nevermind! In the process of recreating this issue from scratch, I discovered it was caused by another dependency pinning `syn` to `1.0.21`. Sorting that out and then running `cargo update -p syn` fixed it.

Sorry for the noise :sweat_smile:

closed 2021-10-05 by antonok-edm