rtic-rs/rtic #545: Suggestion: lock() for an entire context

birkenfeld, 2021-10-16:
The `lock` method on tuples is a great idea for getting multiple resources locked.

However, sometimes I find that I need to lock *all* resources in a `cx.shared` and was wondering if it could also provide a `lock` method that returns a struct with the same field names, but all field tpes are `&mut T`.

Not sure if resources that aren't lockable should be present in that struct or not.

perlindgren, 2021-10-16:
Yes, that should be doable, and quite useful. Thanks for the suggestion we will look into it.

birkenfeld, 2021-10-16:
Nice! I'm really loving the 0.6 changes, btw, it's a big improvement in ergonomics.

perlindgren, 2021-10-17:
There is a crux still to be solved. For soundness, RTIC ensures that you cannot get multiple (mutable) references by locking a field of the shared resources twice. The way its implemented are zero-cost, where the check is done by the compiler at compile time. The resource proxy is consumed by the lock. I'm playing around with this now, if lock is implemented for the "parenting" structure then we should be able to have this working. Some details regarding passing of `cfgs` needs to be ironed out as well.

perlindgren, 2021-10-20:
Small fix, the Priority is no longer leaked. 
https://github.com/rtic-rs/cortex-m-rtic/tree/lockall

birkenfeld, 2021-11-02:
I've tried the current branch and it works fine for me. Thanks!

eflukx, 2022-09-26:
Is there a plan to merge this code?