rtic-rs/rtic #556: nightly for rc?

pdgilbert, 2021-11-16:
`cortex-m-rtic 0.6.0-rc.4`   wants  `rtic-monotonic = "0.1.0-rc.2"` which wants `edition = "2021"` which seems to want me to use nightly.  Is it intentional that I should use nightly for the rc, or am I missing something? I get

```
error: failed to download `rtic-monotonic v0.1.0-rc.2`

Caused by:
  unable to get packages from source

Caused by:
  failed to parse manifest at `/home/paul/.cargo/registry/src/github.com-1ecc6299db9ec823/rtic-monotonic-0.1.0-rc.2/Cargo.toml`

Caused by:
  feature `edition2021` is required

  this Cargo does not support nightly features, but if you
  switch to nightly channel you can add
  `cargo-features = ["edition2021"]` to enable this feature
```

korken89, 2021-11-16:
Hi, if you update the stable compiler to 1.56 it will work fine.

pdgilbert, 2021-11-16:
Thanks. That does it.

closed 2021-11-16 by pdgilbert