rtic-rs/rtic #559: milliseconds with monotonic trait?

pdgilbert, 2021-11-28:
Since the commit `New monotonic trait working` I have been having trouble figuring out how to do a 500 millisecond duration.  I figured out `.seconds()` to `.secs()` but milliseconds still has me stumped. Suggestions would be appreciated.

Also would appreciate a pointer to where/how `. secs()` gets defined since I thought I  might find a clue about milliseconds there, but cannot even figure out how `.secs()` get implemented.

jamwaffles, 2021-11-28:
I think you mean the `secs()` method in [`ExtU32`](https://docs.rs/fugit/latest/fugit/trait.ExtU32.html#tymethod.secs)? That also provides `millis`.

pdgilbert, 2021-12-02:
Thanks @jamwaffles, `millis` was what I needed. And also thanks for the pointer to  [fugit's `ExtU32`](https://docs.rs/fugit/latest/fugit/trait.ExtU32.html#tymethod.secs) for `micros`, `minutes`, `hours`. My `blink_rtic` example is now building again,  CI at https://github.com/pdgilbert/rust-integration-testing/actions/

I'll close this issue, but I am still trying to understand `NOM` and `DENOM` which I seem to need when passing a `Duration` as an argument to a function. I think `NOM` would typically be 1 and `DENOM` would be the clock frequency, but I don't see why they would ever be different from that. A simple explanation or pointer to one would also be appreciated.

closed 2021-12-02 by pdgilbert

burrbull, 2021-12-02:
> I'll close this issue, but I am still trying to understand `NOM` and `DENOM` which I seem to need when passing a `Duration` as an argument to a function. I think `NOM` would typically be 1 and [...]

Just use TimerDuration alias for this

pdgilbert, 2021-12-03:
Thanks @burrbull the TimerDuration  does make it simpler.