rtic-rs/rtic #560: run time spawning problem

pdgilbert, 2021-12-09:
(using git  cortex-m-rtic#2354845 )

I am getting a run time error with task scheduling. The code further below, modified from `example/schedule.rs`, should print

```
init ending
foo
bar
foo
foo
baz
bar
cdr
bar
baz
foo
```
but only prints
```
init ending
foo
bar
cdr
baz
```
The problem seems to be that a task function can only be spawned once. I've tested with `bluepill` and `blackpill stm32f411`.

<details>
<summary>Click to expand code</summary>

```
// from  /cortex-m-rtic/examples/schedule.rs

#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

use panic_semihosting as _;

use rtic::app;

// NOTE: does NOT work on QEMU!

// see <pac>::Interrupt for dispatchers,  e.g. stm32l4::stm32l4x1::Interrupt

#[cfg_attr(feature = "stm32f1xx", app(device = stm32f1xx_hal::pac,   dispatchers = [TIM2, TIM3]))]
#[cfg_attr(feature = "stm32f3xx", app(device = stm32f3xx_hal::pac,   dispatchers = [TIM2, TIM3]))]
#[cfg_attr(feature = "stm32f4xx", app(device = stm32f4xx_hal::pac,   dispatchers = [TIM2, TIM3]))]
#[cfg_attr(feature = "stm32f7xx", app(device = stm32f7xx_hal::pac,   dispatchers = [TIM2, TIM3]))]
#[cfg_attr(feature = "stm32h7xx", app(device = stm32h7xx_hal::pac,   dispatchers = [TIM2, TIM3]))]
#[cfg_attr(feature = "stm32l1xx", app(device = stm32l1xx_hal::stm32, dispatchers = [TIM2, TIM3]))] //fails
#[cfg_attr(feature = "stm32l4xx", app(device = stm32l4xx_hal::pac,   dispatchers = [TIM2, TIM3]))]

//#[rtic::app(device = lm3s6965, dispatchers = [SSI0])]

mod app {
    use cortex_m_semihosting::{hprintln};
    use systick_monotonic::*;

    const MONO_HZ: u32 = 8_000_000; // 8 MHz

    #[monotonic(binds = SysTick, default = true)]
    type MyMono = Systick<100>;

    #[shared]
    struct Shared {}

    #[local]
    struct Local {}

    #[init]
    fn init(cx: init::Context) -> (Shared, Local, init::Monotonics) {
        let systick = cx.core.SYST;

        let mono = Systick::new(systick, MONO_HZ);

        //hprintln!("init").ok();

        //hprintln!("init spawn").ok();
        foo::spawn().ok();
        bar::spawn().ok();
        foo::spawn().ok();

        //hprintln!("init spawn_at").ok();
        bar::spawn_at(monotonics::now() + 10.secs()).ok();
        baz::spawn_at(monotonics::now() + 11.secs()).ok();
        foo::spawn_at(monotonics::now() + 12.secs()).ok();

        //hprintln!("init spawn_after").ok();
        foo::spawn_after(5.secs()).ok();
        baz::spawn_after(6.secs()).ok();
        bar::spawn_after(7.secs()).ok();
        cdr::spawn_after(8.secs()).ok();
        
        hprintln!("init ending").ok();

        (Shared {}, Local {}, init::Monotonics(mono))
    }

    #[task]
    fn foo(_: foo::Context) {
        hprintln!("foo").ok();
    }

    #[task]
    fn bar(_: bar::Context) {
        hprintln!("bar").ok();
    }

    #[task]
    fn baz(_: baz::Context) {
        hprintln!("baz").ok();
    }

    #[task]
    fn cdr(_: cdr::Context) {
        hprintln!("cdr").ok();
    }
}
```
</details>

korken89, 2021-12-09:
Hi,

Your tasks only have s capacity of one so ah ö only the first spawns will work. Replace the ok with unwrape and you'll see it fail.

If you want to queue multiple you need to increase the tasks' capacity :)

pdgilbert, 2021-12-12:
Thanks @korken89. That solves this problem and after some more head banging I have also got my more interesting blink_rtic example working again. I guess simple examples to isolate problems also are good for isolating newbie incompetence.

closed 2021-12-12 by pdgilbert