rtic-rs/rtic #574: Question: specify type of shared resources in my code?

idubrov, 2021-12-29:
Is there any way to specify a type for a shared resource? I want to isolate parts of my code from the whole "context" struct by a subset of it and copying references from full "context". It works fine for local resources (their types are just normal `&mut WhateverMyResourceIs`), but it does not seem like there is a way to do it for shared resources.

For now I use the whole `shared` part:

```rust
pub struct SubsetResources<'a> {
    // ... local resources ... 
    pub shared: crate::app::idle::SharedResources<'a>,
}
```

But I would prefer having separate fields for shared ones. Their internal types seem to be `xyz_that_needs_to_be_locked`, but I cannot reference it from my code since it is private.

Would it be possible to either have a generic type or make generated type referenceable? Ideally, I would like something like this:

```rust
pub struct SubsetResources<'a> {
    // ... local resources ... 
    pub shared: rtic::MutexImpl<'a, WhateverMyResourceIs>,
}
```

korken89, 2022-01-06:
Hi,

You can use the `Mutex` trait and go from there. It should solve your issue.
Mutex trait docs: https://rtic.rs/1.0/api/rtic/trait.Mutex.html

```rust
struct TakesSomeResource<T: Mutex> {
    pub r: T,
}
```

korken89, 2022-01-06:
Here is a working one:

```rust
struct Test<R: Mutex<T = u32>> {
    pub r: R,
}
```

idubrov, 2022-01-06:
If I understand your proposal, this would cause proliferation of type parameters thorough my system (which is what I would want to avoid when possible, just to keep code simpler). I would rather have this struct to have hard-coded field like `some_resource: crate::app::idle::SharedResources_SomeResourceMutex`, with `SharedResources_SomeResourceMutex` being generated by macro & exposed as a public field.

I guess, I could use dynamic dispatch to erase mutex 😈

<details>
  <summary>Don't show this to little kids!</summary>

```rust
trait DynamicMutex<'a, T> {
    fn with_resource(&mut self, cb: &mut dyn FnMut(&mut T));
}

impl <'a, T, M: Mutex<T = T>> DynamicMutex<'a, T> for M {
    fn with_resource(&mut self, cb: &mut dyn FnMut(&mut T)) {
        self.lock(|r| cb(r));
    }
}

pub struct ErasedMutex<'a, T> {
    mutex: &'a mut dyn DynamicMutex<'a, T>,
}

pub trait EraseMutex<'a, T> {
    fn erase(&'a mut self) -> ErasedMutex<'a, T>;
}

impl <'a, T, M: Mutex<T = T>> EraseMutex<'a, T> for M {
    fn erase(&'a mut self) -> ErasedMutex<'a, T> {
        ErasedMutex { mutex: self }
    }
}

impl <'a, T> ErasedMutex<'a, T> {
    pub fn lock<R>(&mut self, cb: impl FnOnce(&mut T) -> R) -> R {
        let mut result: Option<R> = None;
        let mut cb = Some(cb);
        self.mutex.with_resource(&mut |r| {
            result = Some((cb.take().unwrap())(r));
        });
        result.unwrap()
    }
}

struct SomeStruct<'a> {
    something: ErasedMutex<'a, u32>,
}
let x = SomeStruct {
  // `context.shared` is a generated `SharedResources` type
  someting: context.shared.something.erase(),
}

/// Use as normal mutex
x.something.lock(|r| /* ... */);
```
</details>

korken89, 2022-01-06:
Well, that also works :D

closed 2022-01-06 by idubrov