rtic-rs/rtic #575: Problem with dwt-systick-monotonic on 0.6.0 and STM32F103

mryndzionek, 2021-12-29:
I have something like this:
```rs
.
.
.
#[rtic::app(device = stm32f1xx_hal::pac, peripherals = true, dispatchers = [PVD])]
mod app {

    use dwt_systick_monotonic::DwtSystick;
    use glasslcd as _;
    use stm32f1xx_hal::prelude::*;

    #[monotonic(binds = SysTick, default = true)]
    type MyMono = DwtSystick<72_000_000>;
.
.
.
```
and this gives me:
```
5 | #[rtic::app(device = stm32f1xx_hal::pac, peripherals = true, dispatchers = [PVD])]
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `Monotonic` is not implemented for `dwt_systick_monotonic::DwtSystick<72000000_u32>`
```

burrbull, 2021-12-29:
Have you updated `dwt-systick-monotonic` and `cortex-m-rtic` to "1.0.0"?

mryndzionek, 2021-12-29:
> Have you updated `dwt-systick-monotonic` and `cortex-m-rtic` to "1.0.0"?

Yes. I can also confirm that this occurs for STM32F4. If I switch to `0.1.0-rc.2` the issue goes away.

burrbull, 2021-12-29:
Can you show `cargo tree`?

mryndzionek, 2021-12-29:
> Can you show `cargo tree`?
```
├── cortex-m v0.7.3
│   ├── bare-metal v0.2.5
│   │   [build-dependencies]
│   │   └── rustc_version v0.2.3
│   │       └── semver v0.9.0
│   │           └── semver-parser v0.7.0
│   ├── bitfield v0.13.2
│   ├── embedded-hal v0.2.6
│   │   ├── nb v0.1.3
│   │   │   └── nb v1.0.0
│   │   └── void v1.0.2
│   └── volatile-register v0.2.1
│       └── vcell v0.1.3
├── cortex-m-rt v0.6.15
│   ├── cortex-m-rt-macros v0.6.15 (proc-macro)
│   │   ├── proc-macro2 v1.0.36
│   │   │   └── unicode-xid v0.2.2
│   │   ├── quote v1.0.14
│   │   │   └── proc-macro2 v1.0.36 (*)
│   │   └── syn v1.0.84
│   │       ├── proc-macro2 v1.0.36 (*)
│   │       ├── quote v1.0.14 (*)
│   │       └── unicode-xid v0.2.2
│   └── r0 v0.2.2
├── cortex-m-rtic v0.6.0-rc.4
│   ├── bare-metal v1.0.0
│   ├── cortex-m v0.7.3 (*)
│   ├── cortex-m-rtic-macros v0.6.0-rc.4 (proc-macro)
│   │   ├── proc-macro-error v1.0.4
│   │   │   ├── proc-macro-error-attr v1.0.4 (proc-macro)
│   │   │   │   ├── proc-macro2 v1.0.36 (*)
│   │   │   │   └── quote v1.0.14 (*)
│   │   │   │   [build-dependencies]
│   │   │   │   └── version_check v0.9.4
│   │   │   ├── proc-macro2 v1.0.36 (*)
│   │   │   ├── quote v1.0.14 (*)
│   │   │   └── syn v1.0.84 (*)
│   │   │   [build-dependencies]
│   │   │   └── version_check v0.9.4
│   │   ├── proc-macro2 v1.0.36 (*)
│   │   ├── quote v1.0.14 (*)
│   │   ├── rtic-syntax v0.5.0-rc.2
│   │   │   ├── indexmap v1.7.0
│   │   │   │   └── hashbrown v0.11.2
│   │   │   │   [build-dependencies]
│   │   │   │   └── autocfg v1.0.1
│   │   │   ├── proc-macro2 v1.0.36 (*)
│   │   │   ├── quote v1.0.14 (*)
│   │   │   └── syn v1.0.84 (*)
│   │   └── syn v1.0.84 (*)
│   ├── heapless v0.7.9
│   │   ├── hash32 v0.2.1
│   │   │   └── byteorder v1.4.3
│   │   └── stable_deref_trait v1.2.0
│   ├── rtic-core v0.3.1
│   └── rtic-monotonic v0.1.0-rc.2
│   [build-dependencies]
│   └── version_check v0.9.4
├── defmt v0.2.3
│   └── defmt-macros v0.2.3 (proc-macro)
│       ├── defmt-parser v0.2.2
│       ├── proc-macro2 v1.0.36 (*)
│       ├── quote v1.0.14 (*)
│       └── syn v1.0.84 (*)
│   [build-dependencies]
│   └── semver v1.0.4
├── defmt-rtt v0.2.0
│   ├── cortex-m v0.6.7
│   │   ├── aligned v0.3.5
│   │   │   └── as-slice v0.1.5
│   │   │       ├── generic-array v0.12.4
│   │   │       │   └── typenum v1.15.0
│   │   │       ├── generic-array v0.13.3
│   │   │       │   └── typenum v1.15.0
│   │   │       ├── generic-array v0.14.4
│   │   │       │   └── typenum v1.15.0
│   │   │       │   [build-dependencies]
│   │   │       │   └── version_check v0.9.4
│   │   │       └── stable_deref_trait v1.2.0
│   │   ├── bare-metal v0.2.5 (*)
│   │   ├── bitfield v0.13.2
│   │   ├── cortex-m v0.7.3 (*)
│   │   └── volatile-register v0.2.1 (*)
│   └── defmt v0.2.3 (*)
├── dwt-systick-monotonic v1.0.0
│   ├── cortex-m v0.7.3 (*)
│   ├── fugit v0.3.3
│   │   └── gcd v2.1.0
│   │       └── paste v1.0.6 (proc-macro)
│   └── rtic-monotonic v1.0.0
├── heapless v0.7.9 (*)
├── panic-probe v0.2.1
│   ├── cortex-m v0.7.3 (*)
│   └── defmt v0.2.3 (*)
├── rtic-monotonic v1.0.0
└── stm32f1xx-hal v0.7.0 (https://github.com/stm32-rs/stm32f1xx-hal?rev=25702c0#25702c0f)
    ├── bxcan v0.5.1
    │   ├── bitflags v1.3.2
    │   ├── defmt v0.2.3 (*)
    │   ├── embedded-can v0.3.0
    │   │   └── nb v1.0.0
    │   ├── nb v1.0.0
    │   └── vcell v0.1.3
    ├── cast v0.2.7
    │   [build-dependencies]
    │   └── rustc_version v0.4.0
    │       └── semver v1.0.4
    ├── cortex-m v0.6.7 (*)
    ├── cortex-m-rt v0.6.15 (*)
    ├── embedded-dma v0.1.2
    │   └── stable_deref_trait v1.2.0
    ├── embedded-hal v0.2.6 (*)
    ├── nb v0.1.3 (*)
    ├── stm32-usbd v0.6.0
    │   ├── cortex-m v0.7.3 (*)
    │   ├── usb-device v0.2.8
    │   └── vcell v0.1.3
    ├── stm32f1 v0.11.0
    │   ├── bare-metal v0.2.5 (*)
    │   ├── cortex-m v0.6.7 (*)
    │   ├── cortex-m-rt v0.6.15 (*)
    │   └── vcell v0.1.3
    └── void v1.0.2
```

burrbull, 2021-12-29:
cortex-m-rtic v0.6.0-rc.4

Update to 1.0.0

burrbull, 2021-12-29:
And
cortex-m-rt to "0.7"

mryndzionek, 2021-12-29:
cortex-m-rtic is at 1.0.0? Great!! Thanks for the help.

closed 2021-12-29 by mryndzionek