rtic-rs/rtic #576: Dispatcher documentation

ostenning, 2022-01-02:
As a newcomer to this framework, I'm trying to get software tasks to work within my application running on the `stm32h7xx_hal` and I don't fully understand whats going on with Dispatchers and their dependency on the hardware. What are the `SSI0` and `QEI0` symbols?

I think the documentation around this could be improved to help people

korken89, 2022-01-03:
Hi,

Did you read the book on this subject?
https://rtic.rs/1.0/book/en/by-example/software_tasks.html

If so, what do you feel if missing?

korken89, 2022-01-03:
Ah wait, you linked the same page.

> can any unused interrupt vector be used?

Yes.

> Are there any interrupts that generally shouldn't be used
 
No, all vector have settable priority so RTIC can use it as a dispatcher.

> specifically I find this line in the book is a little misleading.

This comes from that you as a user cannot select in which of the provided dispatchers the task will run, RTIC will select one. Software tasks are not directly linked to an interrupt as multiple software tasks are dispatched from a single hardware interrupt.

commit 84a8225d (2022-02-07): Merge #599

599: Docs: SW and HW tasks r=korken89 a=AfoHT

Reword and (hopefully) clarify SW and HW tasks

Closes #576
PR #599 (2022-02-07): Docs: SW and HW tasks
Reword and (hopefully) clarify SW and HW tasks

Closes #576
closed 2022-02-07 by bors[bot]