rtic-rs/rtic #578: Unable to use embedded_time to implement `Monotonic` in 1.0

ZoeyR, 2022-01-04:
Hi,

I can't seem to find a way to implement the `Monotonic` trait using `embedded_time`. The duration and instant types from `embedded_time` don't fulfill the strict requirements put in place in rtic 1.0. Consider the following implementation:
```rust
impl Monotonic for RtcClock {
    type Instant = embedded_time::Instant<Self>;

    type Duration = embedded_time::duration::Microseconds;
}
```
The following error will be produced.
```
error[E0271]: type mismatch resolving `<Instant<RtcClock<R, P>> as Sub>::Output == Microseconds`
   --> embedded-util/src/time.rs:131:5
    |
131 |     type Instant = embedded_time::Instant<Self>;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected struct `Microseconds`, found struct `embedded_time::duration::Generic`
    |
    = note: expected struct `Microseconds`
               found struct `embedded_time::duration::Generic<u64>`
note: required by a bound in `rtic_monotonic::Monotonic::Instant`
   --> /home/zoey/.cargo/registry/src/github.com-1ecc6299db9ec823/rtic-monotonic-1.0.0/src/lib.rs:39:30
    |
39  |         + Sub<Self::Instant, Output = Self::Duration>;
    |
```

However if you switch to using `embedded_time::duration::Generic` you get the following error:

```
error[E0277]: the trait bound `embedded_time::duration::Generic<u64>: FixedPoint` is not satisfied
   --> embedded-util/src/time.rs:131:5
    |
131 |     type Instant = embedded_time::Instant<Self>;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `FixedPoint` is not implemented for `embedded_time::duration::Generic<u64>`
    |
    = note: required because of the requirements on the impl of `Sub<embedded_time::duration::Generic<u64>>` for `Instant<RtcClock<R, P>>`
note: required by a bound in `rtic_monotonic::Monotonic::Instant`
   --> /home/zoey/.cargo/registry/src/github.com-1ecc6299db9ec823/rtic-monotonic-1.0.0/src/lib.rs:38:31
    |
38  |         + Sub<Self::Duration, Output = Self::Instant>
    |
```
Are there any examples of using embedded_time in rtic 1.0? Or do we have to use `fugit` from now on?

korken89, 2022-01-04:
You can use `embedded_time`, but you need to use the [`0.13` release branch](https://github.com/FluenTech/embedded-time/pull/114). 
It implements the requirements we have, and we have tested implementing the `Monotonic` trait against this branch.

ZoeyR, 2022-01-04:
ah, so its not released yet. Good to know.

korken89, 2022-01-04:
Unfortunately not. Maybe worth poking and asking when the release will come.

closed 2022-01-06 by korken89