rtic-rs/rtic #582: app macro doesn't work very well with rust-analyzer

bunnegirl, 2022-01-06:
i've been writing a decently large embedded system to control a dsp and display a vumeter, however i was not particularly happy with how the rust language server integrates with vscode, so i switched to rust analyzer.

the change has been good, and things in general work a lot better, however there's one annoying problem: if i start typing *anything at all* inside of the app macro the entire screen turns red as everything following that attribute macro is considered a compiler error.

[image]

i've not written a proc macro myself, but here the rust analyzer devs talk about how this problem *can* be fixed by the macro developer in at least some cases:

https://github.com/rust-analyzer/rust-analyzer/issues/11014

right now i'm considering externing everything out of the macro just so i can avoid this little problem. which both seems fun, but also extremely pointless.

i thought it worth bringing up in case you were not aware of this and it can be fixed. it seems as tho fixing this in rust analyzer may not be possible in the short term as they have a lot of considerations to make.

perlindgren, 2022-01-06:
Yea,  I saw this problem as well, I thought my `rust-analyzer` was acting up, but it seems to be a more fundamental problem. In general rust analyzer is now better understanding the proc macros, however when there are (syntactic) errors inside the proc macro RTIC + rust analyzer as of today is not able to spot the span of the error (and hence the span will be the entire RTIC module).

I had a quick look at the linked discussion, and it seems that the initial idea was to put the burden on the proc macro implementer (to parse the inner in detail) not sure we want to do that. Right now we just re-emit any tokens that are unknown to RTIC, which  I think is the sensible thing to do atm. In #11014 it was also discussed other potential solutions requiring work on the actual rust-analyzer. So I'm not sure if we want/can do anything about this for the time being. 

I agree however that it is less than optimal that the whole module becomes squiggled while typing, its a lot of noise. 

/Per

korken89, 2022-01-06:
This seems odd, I do not get this problem with rust-analyzer. It correctly highlights the macro error as we would in a build:
[image]

The same with code errors:
[image]

mattico, 2022-01-06:
I'm using this config to reduce the annoyance of this issue: 

```json
"rust-analyzer.procMacro.enable": true,
"rust-analyzer.experimental.procAttrMacros": false,
"rust-analyzer.diagnostics.disabled": [
    "macro-error"
],
```

jonas-schievink, 2022-01-07:
We should definitely shrink the error span here, that can be done without fixing the whole ecosystem

korken89, 2023-04-15:
Closing as the rust analyzer support works quite good now and with the support of having tasks outside the macro eliminates the issue.

closed 2023-04-15 by korken89