rtic-rs/rtic #583: Periodic task execution without jitter / drift

mbrieske, 2022-01-06:
Hi,

in the RTIC 0.5 documentation there is an example in chapter 1.4 "Timer queue" where a task is executed periodically. To prevent delays or jitters in the tasks execution timing, the next execution is spawned inside the task using the contexts field "scheduled":

`cx.schedule.foo(cx.scheduled + PERIOD.cycles()).unwrap();` 

With RTIC v0.6 and the refactoring of task scheduling / spawning it seems that this "scheduled" field was removed from the tasks context. 

Is there now another preferred way of executing periodic tasks that is not mentioned in the docs? Of course I can just manually add a local field, store the execution time there and do stuff manually. That works, but it seems like a step backwards after this functionality had been implemented before? All current examples just call 

`foo::spawn_at(1.secs()).unwrap();`

inside the task method, which does not work for precise timing without drift.

 Thanks

mbrieske, 2022-01-06:
Subsequently, even if manually implementing and updating a "scheduled" field and then calling `foo::spawn_at(scheduled + 1.secs())` instead of `foo::spawn_at(1.secs())`, at the latest when `scheduled + 1.secs()` overflows the timing is corrupted.
When calling `spawn_at` with an argument lower than the current time it seems that the task is executed immediately, so there is no way to schedule the task directly before an overflow occures.
In fact the task is executes multiple times during this short period, because it is always executes immediately.

Here is an output of a task just printing the current time once a second, just before and after the 32 bit monotonic timer overflow happens:

4290000131 us
4291000131 us
4292000131 us
4293000131 us
4294000131 us
4294412555 us
4294794301 us
32835 us
1032835 us
2032835 us
3032835 us
4032835 us

So it seems to me that there is no way to schedule a precisely timed periodic task without using hardware-timers, or am I mistaken?

Thanks

korken89, 2022-01-06:
On the original question, with the new support for multiple monotonics there was no clean way to get the old "scheduled" field so if a user needs this use-case it's up to the user to create a local resource and track the time.

On the second issue, RTIC relies on the underlying time library linked by the `Monotonic` implemented. In this case you seem to be using `fugit`.
Without seeing your code I would guess that there is some error in `fugit`'s time math for overflows, if you can reproduce the error in "raw `fugit`" it would help in the debugging effort.

I'll give it a look, see if I can find the issue.

korken89, 2022-01-06:
Could you also link the complete application code?

perlindgren, 2022-01-06:
Essentially, offsets can relate only to one specific timer. One could think of a trait based extension that a task that is triggered from a timer could use that timer as a "baseline" for future events. However, this needs to be an optional parameter, as all tasks are not time triggered (they can be directly spawned as well, and we don't make any syntactic distinction between time and spawn triggered tasks in RTIC 1.0, in effect the same task could be both time triggered and directly spawned), thus it needs to be optional, if not breaking backwards compatibility with the current API.

korken89, 2022-01-06:
Hi, I have tried to replicate your problem. I used a [1 us counting monotonic](https://github.com/kalkyl/nrf-play/blob/main/src/bin/mono.rs),  and then the following app:

```rust
#![no_main]
#![no_std]

use dwm1001_firmware as _; // global logger + panicking-behavior + memory layout0
pub mod mono;

#[rtic::app(device = dwm1001_firmware::hal::pac, dispatchers = [SWI0_EGU0, SWI1_EGU1])]
mod app {
    use hal::pac::TIMER0;
    use mono::*;
    use rtic::Monotonic;

    #[monotonic(binds = TIMER0, default = true)]
    type Mono = MonoTimer<TIMER0>;

    #[shared]
    struct Shared {}

    #[local]
    struct Local {
        scheduled: <MonoTimer<TIMER0> as Monotonic>::Instant,
    }

    #[init]
    fn init(cx: init::Context) -> (Shared, Local, init::Monotonics) {
        defmt::info!("init");

        let mono = MonoTimer::new(cx.device.TIMER0);

        task1::spawn().ok();

        (
            Shared {},
            Local {
                scheduled: monotonics::now(),
            },
            init::Monotonics(mono),
        )
    }

    #[idle]
    fn idle(_: idle::Context) -> ! {
        defmt::info!("idle");

        loop {}
    }

    #[task(local = [scheduled])]
    fn task1(cx: task1::Context) {
        let now = monotonics::now();
        defmt::info!("Hello from task1! now: {}", now);

        *cx.local.scheduled = *cx.local.scheduled + 1.secs();

        task1::spawn_at(*cx.local.scheduled).ok();
    }
}
```

Which gives the following correct output:

```console
└─ minimal::app::idle @ src/bin/minimal.rs:49
4291000 ms INFO  Hello from task1! now: 4291000009 us
└─ minimal::app::task1 @ src/bin/minimal.rs:57
4292000 ms INFO  Hello from task1! now: 4292000009 us
└─ minimal::app::task1 @ src/bin/minimal.rs:57
4293000 ms INFO  Hello from task1! now: 4293000009 us
└─ minimal::app::task1 @ src/bin/minimal.rs:57
4294000 ms INFO  Hello from task1! now: 4294000009 us
└─ minimal::app::task1 @ src/bin/minimal.rs:57
     32 ms INFO  Hello from task1! now: 32713 us
└─ minimal::app::task1 @ src/bin/minimal.rs:57
   1032 ms INFO  Hello from task1! now: 1032713 us
└─ minimal::app::task1 @ src/bin/minimal.rs:57
   2032 ms INFO  Hello from task1! now: 2032713 us
└─ minimal::app::task1 @ src/bin/minimal.rs:57
   3032 ms INFO  Hello from task1! now: 3032713 us
└─ minimal::app::task1 @ src/bin/minimal.rs:57
   4032 ms INFO  Hello from task1! now: 4032713 us
└─ minimal::app::task1 @ src/bin/minimal.rs:57
   5032 ms INFO  Hello from task1! now: 5032713 us
└─ minimal::app::task1 @ src/bin/minimal.rs:57
```

korken89, 2022-01-06:
Could you share more about what versions of RTIC and monotonics you are using?

mbrieske, 2022-01-06:
Hi, thanks for your many replies, I was not expecting that. I just a week ago started to play around with Rust and RTIC, may very well be that I got something wrong in my code. Here it is:

```rust
use panic_halt as _;

mod mono;

#[rtic::app(device = stm32f7xx_hal::pac, dispatchers = [RCC])]
mod app {
    use super::mono::MonoTimer;
    use cortex_m_semihosting::{hprintln};
    use fugit::{ExtU32, TimerInstantU32};
    use rtic_monotonic::Monotonic;
    use stm32f7xx_hal::{prelude::*, rcc};

    #[monotonic(binds = TIM2, default = true)]
    type MyMono = MonoTimer<1_000_000>;

    #[shared]
    struct Shared {}

    #[local]
    struct Local {
        scheduled: TimerInstantU32<1_000_000>,
    }

    #[init()]
    fn init(cx: init::Context) -> (Shared, Local, init::Monotonics) {
        hprintln!("Init").unwrap();

        let rcc = cx.device.RCC.constrain();
        let ccdr = rcc
            .cfgr
            .hse(rcc::HSEClock::new(8.MHz(), rcc::HSEClockMode::Oscillator))
            .use_pll()
            .sysclk(216.MHz())
            .freeze();

        let mut mono = MyMono::new(cx.device.TIM2, &ccdr);

        run::spawn_after(1.seconds()).unwrap();

        (Shared {}, Local {scheduled: mono.now()}, init::Monotonics(mono))
    }

    #[task(local = [scheduled])]
    fn run(cx: run::Context) {
        let mut scheduled = *cx.local.scheduled;
        hprintln!("{}", monotonics::now()).unwrap();
        scheduled = scheduled + 1.secs();
        *cx.local.scheduled = scheduled;
        run::spawn_at(scheduled).unwrap();
    }
```

As for the monotonic timer, I implemented my own based on the example implementation mentioned in the documentation for the STM32F411 (https://github.com/kalkyl/f411-rtic/blob/main/src/bin/mono.rs). I pretty much copied it with some minor changes to make it work on the STM32F765. I can of course post the code here if it helps. 

For testing this issue I realize that a tick period of 1ms like I have in my code might be a little slow, I just copied my code regardless just to not change anything that could relate to this bug.

Thanks!

mbrieske, 2022-01-06:
Ah I can just post the code for the monotonic here regardless. Maybe it helps.

```rust
use rtic_monotonic::Monotonic;
use stm32f7xx_hal::{pac, rcc};

pub struct MonoTimer<const FREQ: u32> (pac::TIM2);

impl<const FREQ: u32> MonoTimer<FREQ> {
    pub fn new(tim2: pac::TIM2, clocks: &rcc::Clocks) -> Self {
        let rcc = unsafe { &(*pac::RCC::ptr())};
        rcc.apb1enr.modify(|_, w| w.tim2en().set_bit());
        rcc.apb1rstr.modify(|_, w| w.tim2rst().set_bit());
        rcc.apb1rstr.modify(|_, w| w.tim2rst().clear_bit());
        let prescaler = clocks.pclk2().0 / FREQ -1;
        tim2.psc.write(|w| w.psc().bits(prescaler as u16));
        tim2.arr.write(|w| unsafe { w.bits(u32::MAX) });
        tim2.egr.write(|w| w.ug().set_bit());
        tim2.sr.modify(|_, w| w.uif().clear_bit());
        tim2.cr1.modify(|_, w| w.cen().set_bit().udis().set_bit());
        MonoTimer(tim2)
    }
}

impl<const FREQ: u32> Monotonic for MonoTimer<FREQ> {
    type Instant = fugit::TimerInstantU32<FREQ>;
    type Duration = fugit::TimerDurationU32<FREQ>;

    #[inline(always)]
    fn now(&mut self) -> Self::Instant {
        Self::Instant::from_ticks(self.0.cnt.read().cnt().bits())
    }

    fn set_compare(&mut self, instant: Self::Instant) {
        self.0
            .ccr1
            .write(|w| w.ccr().bits(instant.duration_since_epoch().ticks()));
    }

    fn clear_compare_flag(&mut self) {
        self.0.sr.modify(|_, w| w.cc1if().clear_bit());
    }

    #[inline(always)]
    fn zero() -> Self::Instant {
        Self::Instant::from_ticks(0)
    }

    unsafe fn reset(&mut self) {
        self.0.dier.modify(|_, w| w.cc1ie().set_bit());
    }
}
```

korken89, 2022-01-06:
Thank you! Can you also paste your `Cargo.toml`?

mbrieske, 2022-01-06:
> Thank you! Can you also paste your `Cargo.toml`?

Sure:

```
[package]
name = "cmcu"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
fugit = "0.3.0, < 0.3.3"
cortex-m = "0.6.0"
cortex-m-rt = "0.6.10"
cortex-m-semihosting = "0.3.3"
cortex-m-rtic = "1.0.0"
panic-halt = "0.2.0"
rtic-monotonic = "1.0.0"
embedded-time = "0.12.1"

#[dependencies.stm32f7]
#version = "0.14.0"
#features = ["stm32f765", "rt"]

[dependencies.stm32f7xx-hal]
version = "0.6.0"
features = ["stm32f765", "rt"]

[[bin]]
name = "test"
test = false
bench = false

[profile.release]
codegen-units = 1 # better optimizations
debug = true # symbols are nice and they don't increase the size on Flash
lto = true # better optimizations
```

korken89, 2022-01-06:
Could you try updating to the latest `fugit`?
There were a lot of fixes from 0.3.0, see if it helps.

mbrieske, 2022-01-06:
Oh man. Using fugit 0.3.2 instead of 0.3.0 works.

I could not use the latest version, fugit 0.3.3, because it will not compile on my machine with this configuration - thats why I just used the 0.3.0 version. See the attached error message, but I will probably just create a ticket in the fugit github.

```error[E0658]: panicking in constants is unstable
  --> /home/mbrieske/.cargo/registry/src/github.com-1ecc6299db9ec823/fugit-0.3.3/src/helpers.rs:48:32
   |
48 |     pub const GREATER_EQ: () = assert!(L >= R);
   |                                ^^^^^^^^^^^^^^^
   |
   = note: see issue #51999 <https://github.com/rust-lang/rust/issues/51999> for more information
   = note: this error originates in the macro `assert` (in Nightly builds, run with -Z macro-backtrace for more info)
```

Great find @korken89, really nailed it. Thanks!

Cheers

korken89, 2022-01-06:
Awesome, glad you got it working!

To fix your `fugit` problem, update to the latest stable Rust compiler. It requires 1.57. :)

closed 2022-01-06 by korken89