rtic-rs/rtic #586: Access to mutable references from tasks via Shared

mryndzionek, 2022-01-18:
In `init` I have:
```rs
let mut flash = c.device.FLASH.constrain()
```
Then I have a module that requires mutable reference to `flash` and need to put it into `Shared`. Is there a way to accomplish this without using `unsafe`? It seems that there isn't, as I would need some lifetime in `Shared` to attache the reference to.

If there isn't, is doing (unsafe) `stm32f1xx_hal::pac::Peripherals::steal()` in tasks an acceptable solution?

perlindgren, 2022-01-19:
Not sure how your use case looks like, can you paste the relevant parts here and we can have a look.
/Per

korken89, 2022-01-19:
Hi, for this you need to promote the reference to a static lifetime.
It can be done like this:

```rust
#[init(local = [flash: Option<FLASH> = None;]]
fn init(cx: init::Context) {
    // Make the flash static and get a 'static reference
    let flash = cx.local.flash.insert(cx.device.FLASH);

    // Create your thing that now has 'static lifetime
    let mut flash = flash.constrain();

    // ...
}
```

mryndzionek, 2022-01-19:
Unfortunately still doesn't work:
```rs
let mut flash = flash.constrain();
    |                         ^^^^^^^^^^^^^^^^^ move occurs because `*flash` has type `stm32f1xx_hal::pac::FLASH`, which does not implement the `Copy` trait
```

korken89, 2022-01-19:
Huh then it's not working on references it seems. It should just work then by letting flash take ownership of the hardware peripheral.
But at least you know how to generate a static lifetime if needed.

korken89, 2022-01-19:
I gave the HAL a check and it takes the peripheral by move, so you don't need to work with references.

mryndzionek, 2022-01-19:
Yes! This works! Thank you!

closed 2022-01-19 by mryndzionek