rtic-rs/rtic #587: [Book] Make it easy to follow along with the examples

GreenPenguino, 2022-01-18:
It would be nice to add instructions on how to follow along with the 1.0 version of the book. At the moment it just points to The embedded rust book.

I followed all the examples from The embedded rust book previously, so I have all the tools installed. However, just using the [cortex-m-quickstart](https://docs.rust-embedded.org/book/start/qemu.html) and pasting the code of the [first example](https://github.com/rtic-rs/cortex-m-rtic/blob/master/examples/common.rs) to main.rs will not work out of the box. It leaves me to solve all the compile errors caused by missing packages and unstable features.

It would be nice to have a similar template as cortex-m-quickstart for rtic, and/or a bit more elaborate instructions on what to do if you want to follow along with the examples.

Some of the issues are:
- missing `panic-semihosting` in `Cargo.toml`
- missing `systick-monotonic` in `Cargo.toml`
- panicking in constants is unstable (currently stuck here)

perlindgren, 2022-01-19:
Hi

We have a similar template for RTIC. https://github.com/rtic-rs/defmt-app-template

In general it should be no need for unstable features, RTIC as such compiles and runs under stable rust since a long time.

/Per

korken89, 2022-01-19:
Hi, the `panic` you are getting is because you use an old compiler.
Update to the latest `stable` and it will work.

GreenPenguino, 2022-01-19:
@korken89 You are right, I updated from 1.56 to 1.58 and I don't get that warning anymore. 

Now I have issues with the device in the example, lm3s6965. The trait `core::marker::Copy` is not implemented for `Interrupt`.

AfoHT, 2022-01-19:
Regarding the book, the book refers to the template suggested by @perlindgren, but if you have further suggestions we're all ears

https://rtic.rs/1.0/book/en/by-example/starting_a_project.html

GreenPenguino, 2022-01-19:
Okay, now it starts to make sense. I was reading the book in sequential order, so I didn't get to that chapter yet. 
I think it would be wise to mention that page in [1. RTIC by example](https://rtic.rs/1.0/book/en/by-example.html)

AfoHT, 2022-01-21:
I see. Surely there are many ways of approaching this reading material, but for my understanding of your approach, you've already read the Embedded Rust book, tried `cortex-m-quickstart` at some point, and then tried to combine `examples/common.rs` into a new "project" created with quickstart?

All examples in `examples/` folder come prepared to run with QEMU, the template linked is not bound to any specific hardware, to make it QEMU runnable requires the users to know which dependencies to add, so maybe not the first step.

The idea with that template is to have hardware agnostic but still desirable components "pre-packaged", naturally that still requires some DIY assembly :D 

RTIC v1.0 is `edition = "2021"`, meaning >= 1.56, for more details see [announcing Rust Edition 2021][2021]

[2021]: https://blog.rust-lang.org/2021/10/21/Rust-1.56.0.html

GreenPenguino, 2022-01-22:
@AfoHT Yes, that is right. However, running the lm3s6965 device in Qemu requires the right dependency, since RTIC needs those dispatcher names. In The embedded rust book, it was not necessary to add a dependency related to lm3s6965, since no device-specific features are used there. Which dependency needs to be added to make the lm3s6965 example work in Qemu, because I was not able to figure it out. The crate [lm3s6965](https://crates.io/crates/lm3s6965) did not do the trick for me.

And about the Rust version required, [the book](https://rtic.rs/1.0/book/en/#requirements) states that the 2018 version is used. Is the book outdated?

AfoHT, 2022-01-22:
I see. The examples uses [Cargo.toml](https://github.com/rtic-rs/cortex-m-rtic/blob/master/Cargo.toml) and in the end of the file a patched version is used, I suggest you do the same.

About the edition, you are correct, this is an error and needs an update in the book, thanks for pointing it out!

If you'd like this is an excellent first issue to tackle

commit 69cacd05 (2022-01-22): Merge #588

588: Highlight how to run examples locally r=perlindgren a=AfoHT

How examples are run never gets properly introduced before used in `by-example/resources.md`

Fixes #587
PR #588 (2022-01-22): Highlight how to run examples locally
How examples are run never gets properly introduced before used in `by-example/resources.md`

Fixes #587
closed 2022-01-22 by bors[bot]