rtic-rs/rtic #590: CI for book broken

GreenPenguino, 2022-01-23:
The Github action for deploying the book does not seem correctly configured. The latest changes on the `master` branch are not visible on [rtic.rs](https://rtic.rs).

Take for example [Merge #588](https://github.com/rtic-rs/cortex-m-rtic/pull/588), which triggered [pages build and deployment](https://github.com/rtic-rs/cortex-m-rtic/actions/runs/1734213904). The changes are not visible on [the page in the book](https://rtic.rs/1.0/book/en/by-example.html)

AfoHT, 2022-01-24:
The latest changes on `master` branch are visible on [rtic.rs/dev](https://rtic.rs/dev/)

They are also linked to in the "Preface" with the title "Development version"

The idea is that rtic.rs will point to latest stable, in this case rtic.rs/v1.0, which builds from branch `v1.0.x` and allow development builds to not break stable docs :)

GreenPenguino, 2022-01-24:
So it will take a while for fixes in the book like #588 and #591 will make it to [rtic.rs](https://rtic.rs)?

AfoHT, 2022-01-26:
Depends on how you see it, #588 is currently visible here: https://rtic.rs/dev/book/en/by-example.html

[rtic.rs/1.0/](https://rtic.rs/1.0/book/en/) will be updated when a new release is pushed to the `1.0.x` branch, uncertain exactly when.

closed 2022-02-05 by AfoHT