rtic-rs/rtic #611: Naming lock types

LU15W1R7H, 2022-02-15:
I want to pass a shared resource to a function, but I'm unable to name the lock.
The path `app::shared_resources::resource_that_needs_to_be_locked` is not accessible, as `app::shared_resources` is private.
Also the generated name `resource_that_needs_to_be_locked` is inconvenient.

mattico, 2022-02-15:
Does passing it using the `Mutex` trait work? e.g. `fn do_work(counter: impl rtic::Mutex<u32>)`

LU15W1R7H, 2022-02-15:
Unfortunately no, because it's a trait object function.

LU15W1R7H, 2022-02-15:
@AfoHT Sure!

This doesn't work...
```rust
#[derive(Default)]
struct Resource(u32);

trait MyTrait {
    fn my_func(&self, resource_lock: impl Mutex<T = Resource>);
}
struct MyStruct;
impl MyTrait for MyStruct {
    fn my_func(&self, resource_lock: impl Mutex<T = Resource>) {
        resource_lock.lock(|r| r.0 = 1);
    }
}

fn check() {
    let resource = Resource::default();
    let trait_object: &dyn MyTrait = &resource;
}
```

But with #612 this works!
```rust
#[derive(Default)]
struct Resource(u32);

trait MyTrait {
    fn my_func(&self, resource_lock: &mut app::shared_resources::resource_lock);
}
struct MyStruct;
impl MyTrait for MyStruct {
    fn my_func(&self, resource_lock: &mut app::shared_resources::resource_lock) {
        resource_lock.lock(|r| r.0 = 1);
    }
}

fn check() {
    let resource = Resource::default();
    let trait_object: &dyn MyTrait = &resource;
}
```

Or do you want me to craft a minimal project in a repository? It would be using the `rp_pico` board crate.

LU15W1R7H, 2022-02-15:
@AfoHT 
Sure! I created a git repository with two branches!

- [Problem branch](https://github.com/LU15W1R7H/rtic-issue-611/tree/problem)
- [Fix branch](https://github.com/LU15W1R7H/rtic-issue-611/fix/problem)

It uses #612.

perlindgren, 2022-02-15:
Is it when you want to embed the function into a trait that the problem occurs. I mean in general you can pass a `Mutex<T>` to a free function like this: 

https://github.com/rtic-rs/cortex-m-rtic/blob/master/examples/generics.rs

LU15W1R7H, 2022-02-15:
@perlindgren Yes, it's about trait object safety!