rtic-rs/rtic #615: Syntax highlighting Clion

Ridtt, 2022-02-18:
Using stm32f103 example project [-](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/examples/timer-interrupt-rtic.rs) projects builds just fine,  but I dont have any syntax highlighting inside functions such as 

`
#[task(binds = TIM1_UP, priority = 1, local = [led, timer_handler, led_state: bool = false, count: u8 = 0])]
fn tick(cx: tick::Context) { /*...*/ }
`
every type macro brings is unknow.

using latest Clion ( 2021.3.3) with rust plugin (0.4.165.4438-213)

Assuming everything is powered by macro, this is unusable in curent state.
Im planing on huge RT system on RTIC for a few weak and I really stuck now and cant go even step forward before i can solve this issue. Please help.

Ridtt, 2022-02-18:
By enabling experemental fuatures from
> [clion procedural macro support](https://github.com/intellij-rust/intellij-rust/issues/6908#issue-818633667) 
im getting sytax highlighting. But sad thing is - when i start typing anything its recompiles macro, it's become invalid and highlighting is turning off. This way you can read code, but can't write using autocomplete. Need to find more information.

perlindgren, 2022-02-18:
Yea, its a bit tricky since the proc-macro generates code on the fly that the IDE needs to be aware of for completions etc. Using Rust analyzer this now works as long as the code within the proc-macro (your app) is correct, else it may abort early. Also error spans might be too large in cases (covering the whole `app`). So still Rust analyzer support is not perfect, but its getting there.

A suggestion is to make the `app` minimal and use external implementations for tasks. 

/Per

Ridtt, 2022-02-18:
@korken89 @perlindgren 
You can close this if nothing else could be done at this moment or leave open for someone else suggestions. Up to you guys. Thanks.

perlindgren, 2022-02-19:
Closing for now.

closed 2022-02-19 by perlindgren