rtic-rs/rtic #634: Strange compilation error while migrating to RTIC v1

TeXitoi, 2022-04-14:
I've tried to migrate to rtic v1 here: https://github.com/TeXitoi/keyseebee/tree/rtic1 but I have a strange error:

```
$ cargo build --release 
   Compiling keyseebee v0.1.0 (.../keyseebee/firmware)
error[E0080]: evaluation of constant value failed
  --> src/main.rs:42:1
   |
42 | #[app(device = crate::hal::pac, peripherals = true, dispatchers = [CEC_CAN])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ attempt to compute `4_usize - 5_usize`, which would overflow
   |
   = note: this error originates in the attribute macro `app` (in Nightly builds, run with -Z macro-backtrace for more info)

For more information about this error, try `rustc --explain E0080`.
error: could not compile `keyseebee` due to previous error
```

I've tried to force using the v1.0.0 version, but I have the same error:

```
$ git diff Cargo.toml
diff --git a/firmware/Cargo.toml b/firmware/Cargo.toml
index 8026e51..c3b6091 100644
--- a/firmware/Cargo.toml
+++ b/firmware/Cargo.toml
@@ -10,11 +10,15 @@ cortex-m = "0.7"
 cortex-m-rt = { version = "0.7.1", features = ["device"] }
 panic-halt = "0.2.0"
 keyberon = { git = "https://github.com/TeXitoi/keyberon", branch = "master" }
-cortex-m-rtic = "1.1.1"
+cortex-m-rtic = "=1.0.0"
+cortex-m-rtic-macros = "=1.0.0"
 embedded-hal = "0.2"
 usb-device = "0.2.0"
 nb = "1.0.0"
 
+[dev-dependencies]
+rtic-syntax = "=1.0.0"
+
 [profile.release]
 lto = true
 incremental = false
$ cargo outdated
    Updating git repository `https://github.com/TeXitoi/keyberon`
Name                                 Project  Compat  Latest  Kind         Platform
----                                 -------  ------  ------  ----         --------
cortex-m-rtic                        1.0.0    ---     1.1.1   Normal       ---
cortex-m-rtic->cortex-m-rtic-macros  1.0.0    ---     1.1.0   Normal       ---
cortex-m-rtic-macros                 1.0.0    ---     1.1.0   Normal       ---
cortex-m-rtic-macros->rtic-syntax    1.0.0    ---     1.0.1   Normal       ---
rtic-syntax                          1.0.0    ---     1.0.1   Development  ---
```

TeXitoi, 2022-04-14:
> For Cortex-M, tasks can have priorities in the range 1..=(1 << NVIC_PRIO_BITS) where NVIC_PRIO_BITS is a constant defined in the device crate.

OK, I have NVIC_PRIO_BITS = 2, so 4 is the maximum priority, and I had a priority of 5.

Could be great if the error was more readable.

AfoHT, 2022-04-19:
Let's reopen this to track that we need to improve the error message, thank you for reporting!

commit 1a24c725 (2022-05-17): Merge #641

641: More ergonomic error from static asserts messages r=perlindgren a=korken89

Closes #634
PR #641 (2022-05-17): More ergonomic error from static asserts messages
Closes #634
closed 2022-05-17 by bors[bot]