rtic-rs/rtic #639: splitting app source code in multiple files

lelongg, 2022-05-11:
Is there a way to split the app module source code into different files ? Each of my attempts feels like hurting some wall. The main issue seems to be that having a non-inline module inside a macro is currently unstable. Is everybody else keeping the whole app in a single file ?

perlindgren, 2022-05-11:
You can implement tasks external to the app macro. (I think for now init/idle needs to be internal in the app, eventually we will support all task types, there is a PR on the way, but some implementation work remains...)

/Per

eflukx, 2022-07-18:
Hmm.. been fighting with this for some time.. but I really don't seem to be able to access the RTIC main `app` module, which is defined in main.rs (or in my case an example.rs) main binary, from outside of the `app` module.

The `extern_spawn.rs` does show an external spawn for a task function, but the code is still within the same file. It is probably something trivial, but how make the 'main' `app` module `pub` to allow it to be `use`d somewhere else?

sakian, 2022-07-18:
In my secondary file I have `use crate::app;`. I can then spawn tasks within my extern function using `app::<task_name>::spawn().unwrap();`.

eflukx, 2022-07-19:
Hi @sakian thanks for your response on this (long dead) topic!

I got things working (more or less) the way I desire. In my app I have `main.rs` and a `lib.rs`, where the latter is referencing most of the app's modules. Apparently rust/cargo compiles `main.rs` and `lib.rs` a separate crates. Up until now stuff like modules and `use`-ing code (from internal and external sources), just worked™️, but RTIC with external binding does not seem to fit this nicely. (as the app 'root' needs to be in the 'executable' crate and can not be in the 'lib part').

When I chuck all the `pub mod xxx` from `lib.rs` into the main.rs things start working as expected. It's just not possible (or at least I don't know how) to use (reference) the main crate from the lib crate. Currently I placed all the mods that have to do with the RITC app in the main.rs file and that's fine for now. 

Another thing I noticed is that having a qualified/mod name in the extern block's function name, seems to break things. This does not work: ( One has to `use mymod::foo` and use the short name)

```rust
    extern "Rust" {
        #[task()]
        fn mymod::foo(_c: foo::Context, _x: i32, _y: u32);
    }
```

All in all I got working what was needed, so that's good. Some more elaborate (multi file) examples resembling a "real" application would be a welcome addition for RTIC imho..