rtic-rs/rtic #642: RTIC introduces ambiguous assert!

evils, 2022-05-23:
i'm using `defmt::*` which imports `defmt::assert`

since my last `cargo update` (`cortex-m-rtic-macros v1.0.0 -> v1.1.2`)
a conflict with another `assert` macro was introduced
[image]

switching to `use defmt::{trace, debug}` resolved this
(i was already using `defmt::assert!` explicitly, possibly for a similar problem)

the RTIC matrix channel suggested i look at `rtic-expansion.rs`
here is the snippet from there i believe to conflict with `defmt::assert` (via `defmt::*`)
```
{
    if rtic :: export :: is_armv6()
    {
        assert!
        ((stm32f4xx_hal :: pac :: Interrupt :: OTG_FS as u32) < 32)
        ;
    } else {}
} ;
```
@jamesmunns suggested that this problem could be avoided if RTIC namespaced its use of the assert macro

jamesmunns, 2022-05-23:
It looks like this was introduced in https://github.com/rtic-rs/cortex-m-rtic/pull/589, in the `1.1.2` release.

evils, 2022-05-23:
note that the changes on `master` since `1.1.2` also introduce a conflict with `defmt::panic`

perlindgren, 2022-05-23:
Good catch, we will look into it.
/Per

korken89, 2022-05-24:
@evils I checked the codegen and I could only find `assert!` that were colliding.
I was not able to find any `panic!` at all in the codegen. Where did you find the offending `panic!`?

evils, 2022-05-24:
i also have no idea where the panic is
[image]

do note i have basically this in my project, which does define a `fn panic`
https://github.com/knurling-rs/app-template/blob/main/src/lib.rs

(side note, screenshot is with your PR, it does resolve the assert conflict :) )

korken89, 2022-05-24:
@evils Hmm, I built the template but can't find any panic.
Can you have a look in the `target/rtic-expansion.rs` file and print screen any panic?
This file holds the expanded RTIC code after codegen.

As long as we see it it context it should be easy to fix. :)

korken89, 2022-05-24:
Wait what, your error says that it originates from inside `::core::assert`.

evils, 2022-05-24:
oh, apparently `rg` ignores `target/`
but `rtic-expansion.rs` does not contain the string `panic`

evils, 2022-05-24:
in case it's of use, my project is public
a `cargo update` at this point should yield the original error reported here
https://gitlab.com/evils/ASC/-/tree/5df5df8ba3846282f96dcd0ea713017c6b0969fe

and pointing `cortex-m-rtic` to the current master branch or your PR's branch,
should incur both or just the panic error, respectively

korken89, 2022-05-24:
Hi, I gave it a try and can reproduce it.
The odd thing is that there is no `panic` in the code - it seems that the `assert!` macro expansion internally uses `panic!` instead of `::core::panic!` which causes the error.

I have switched in the codegen to not use `assert!` which solves the issue.
However this is a `core` lib bug...

commit 5fe63502 (2022-05-24): Merge #643

643: Fix clash with defmt r=AfoHT a=korken89

Fixes #642
PR #643 (2022-05-24): Fix clash with defmt
Fixes #642
closed 2022-05-24 by bors[bot]