rtic-rs/rtic #654: Cfg blocks in Shared Resource fields

David-OConnor, 2022-07-09:
Code to replicate:
```rust
    #[shared]
    struct Shared {
        #[cfg(feature = "h7")]
        uart_elrs: Usart<USART7>,
        #[cfg(feature = "g4")]
        uart_elrs: Usart<USART3>
    }
```

Error:
```rust
error: this resource is listed more than once
   --> src\main.rs:310:9
    |
310 |         uart_elrs: Usart<USART3>,
    |         ^^^^^^^^^
```

For normal struct fields, the syntax I posted is valid. I suspect this has to do with RTIC being macro-based.

korken89, 2022-08-02:
Indeed, it's a long standing issue to have `cfg` support.
In a nutshell the issue is that we can't analyze if the `cfg` is active or not, so we need to generate code assuming both are active.
This causes clashes in many cases, as you have noticed. 

So the general recommendation for now is, don't use `cfg` in RTIC.
Instead, move the `cfg` to a user type wrapping the thing you want to have in RTIC.

korken89, 2022-09-21:
Closing, feel free to reopen if you have more questions.

closed 2022-09-21 by korken89

David-OConnor, 2022-09-21:
Why is it closed /closed as completed? Is this a won't-fix?