rtic-rs/rtic #655: Problem with passing `static` locals

mryndzionek, 2022-07-13:
In my task I have a type that requires explicit lifetime. The only thing (I think) I can do is to use `'static`. Then I would like to avoid `unsafe` code, so I'm trying to pass a mutable reference to the needed static from `init` (essentially "static mut transform"). It all works in `init`, but when I try to use/access the `static` reference via `Context` in the target task I get:
```
`cx` has an anonymous lifetime `'_` but it needs to satisfy a `'static` lifetime requirement
``` 
Is there a way to solve this?

korken89, 2022-08-02:
Hi, could you share the code?
Without seeing it, it's difficult to give good advice.

However generally, there should be no problem generating `'static` lifetime objects in `init`.
You can also have a look here: https://rtic.rs/1/book/en/by-example/tips_static_lifetimes.html

mryndzionek, 2022-08-02:
The code is fairly long and I don't have time to simplify, but essentially I have a type like:
```rs
type Xfer = Transfer<stm32f1xx_hal::dma::R, &'static [u8; LEDS_BUF_SIZE], Leds>
```
If I try to use a variable of this type in a task it doesn't work. It only works in `init`. I've made a workaround for now. I wrapped the variable in an `Option`, I have a static buffer inside the task and then I need to resort to on small `unsafe`.

korken89, 2022-09-21:
Closing, if there are any more questions feel free to reopen.

closed 2022-09-21 by korken89