rtic-rs/rtic #658: not enough interrupts to dispatch all software tasks

adminSxs, 2022-09-05:
```rust

#![no_std]
#![no_main]

use defmt_rtt as _; 
use panic_probe as _;

#[rtic::app(device = stm32f1xx_hal::pac)]
mod app {
    use defmt::println;

    #[shared]
    struct Shared {}

    #[local]
    struct Local {}

    #[init]
    fn init(cx: init::Context) -> (Shared, Local, init::Monotonics) {
        println!("init ...");

        (Shared {}, Local {}, init::Monotonics())
    }

    #[idle]
    fn idle(cx: idle::Context) -> ! {
        println!("continue");
        bar::spawn().unwrap();
        loop {
            continue;
        }
    }

    #[task]
    fn bar(_: bar::Context) {
        println!("bar");
    }
}

````
error: not enough interrupts to dispatch all software tasks (need: 1; given: 0)

burrbull, 2022-09-05:
Answer in the book https://rtic.rs/1/book/en/by-example/software_tasks.html . See dispatchers

If you want to ask question you could do it in chat instead of open issue:
https://matrix.to/#/#rtic:matrix.org
https://matrix.to/#/#rust-embedded:matrix.org

perlindgren, 2022-09-14:
I think we can close this issue, please re-open if needed.

closed 2022-09-14 by perlindgren