rtic-rs/rtic #662: Local resources used in #[cfg]-disabled tasks incorrectly considered shared

little-arhat, 2022-10-23:
Modifying `cfg-whole-task` example from the repo:

```lang:rust

//! examples/cfg-whole-task.rs

#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

use panic_semihosting as _;

#[rtic::app(device = lm3s6965,  dispatchers = [SSI0, QEI0])]
mod app {
    use cortex_m_semihosting::debug;
    #[cfg(debug_assertions)]
    use cortex_m_semihosting::hprintln;

    #[shared]
    struct Shared {
        #[cfg(never)]
        unused: u32,
    }

    #[local]
    struct Local {
        count: u32,
    }

    #[init]
    fn init(_: init::Context) -> (Shared, Local, init::Monotonics) {
        foo::spawn().unwrap();
        foo::spawn().unwrap();

        (
            Shared {
                #[cfg(never)]
                unused: 1,
            },
            Local {
                count: 0
            },
            init::Monotonics(),
        )
    }

    #[idle]
    fn idle(_: idle::Context) -> ! {
        debug::exit(debug::EXIT_SUCCESS); // Exit QEMU simulator

        loop {
            cortex_m::asm::nop();
        }
    }

    #[task(capacity = 2, local = [count])]
    fn foo(mut _cx: foo::Context) {
        #[cfg(debug_assertions)]
        {
            _cx.shared.count.lock(|count| *count += 1);

            log::spawn(_cx.shared.count.lock(|count| *count)).unwrap();
        }

        // this wouldn't compile in `release` mode
        // *_cx.shared.count += 1;

        // ..
    }

    // The whole task should disappear,
    // currently still present in the Tasks enum
    #[cfg(never)]
    #[task(capacity = 2, local = [count])]
    fn foo2(mut _cx: foo2::Context) {
        #[cfg(debug_assertions)]
        {
            _cx.shared.count.lock(|count| *count += 10);

            log::spawn(_cx.shared.count.lock(|count| *count)).unwrap();
        }

        // this wouldn't compile in `release` mode
        // *_cx.shared.count += 1;

        // ..
    }

    #[cfg(debug_assertions)]
    #[task(capacity = 2)]
    fn log(_: log::Context, n: u32) {
        hprintln!(
            "foo has been called {} time{}",
            n,
            if n == 1 { "" } else { "s" }
        )
        .ok();
    }
}
```

and trying to compile:

```
>>= cargo build --example cfg-whole-task --target thumbv7em-none-eabihf
   Compiling cortex-m-rtic v1.1.3 (/Users/r/prj/p/copterust/japaric/cortex-m-rtfm)
error: Local resource "count" is used by multiple tasks or collides with multiple definitions
  --> examples/cfg-whole-task.rs:53:35
   |
53 |     #[task(capacity = 2, local = [count])]
   |                                   ^^^^^

error: Local resource "count" is used by multiple tasks or collides with multiple definitions
  --> examples/cfg-whole-task.rs:71:35
   |
71 |     #[task(capacity = 2, local = [count])]
   |                                   ^^^^^

error: could not compile `cortex-m-rtic` due to 2 previous errors
```

Even thoug `foo2` is behind `cfg(never)`.

Expected behaviour: code is accepted with local resource.

perlindgren, 2022-10-23:
I haven't looked into this specific example, but in general its not possible for RTIC to evaluate (any) cfg, so the analysis need to be pessimistic (i.e. assume that the task will be there). The only thing we can currently do regarding cfg:s is to re-emit them during synthesis, so that the Rust compiler (later) evaluates the macro and removes the code. 

That behavior is not specific to RTIC but a limitation of the current proc-macro design in Rust. 
/Per

perlindgren, 2022-11-02:
I'll close this for now, feel free to re-open if you still have questions.

closed 2022-11-02 by perlindgren

little-arhat, 2022-11-12:
@perlindgren well, crate announces some support for `#[cfg]`, there is even a test. Support is just far from complete. While this task can be worked around (by using "shared" resouces), related bug is more annoying: https://github.com/rtic-rs/cortex-m-rtic/issues/665