rtic-rs/rtic #663: simplify binding tasks to EXTI

rursprung, 2022-10-30:
i've tried out RTIC for the first time on a Nucleo F401RE development board with a simple adaptation of [this example](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/rtic.rs) from `stm32f4xx-hal`. the only change i did was to use `PA5` for the LED and `PC13` for the button (because that's the placement for the built-in LED & button on this board).

and then the fun started: the program started fine, but the button click would not be registered. it took me a while to find out that for `PC13` it wasn't `EXTI0` but instead `EXTI15_10` (i'm pretty new to STM32 & embedded in general) - i didn't even find this in the datasheet of the microcontroller.

i also checked your example, which also used a pin which used `EXTI0`:
https://github.com/rtic-rs/rtic-examples/blob/6f4a07f5beb68381b801ca30bf05114b31a831a9/rtic_v1/stm32f411_edge_counter/src/main.rs#L66

the only way i found the mapping was by looking at [the implementation of `Pin::interrupt`](https://github.com/stm32-rs/stm32f4xx-hal/blob/6d0c29233a4cd1f780b2fef3e47ef091ead6cf4a/src/gpio/exti.rs#L8-L23) which contains the mapping.

i think it'd be great if the API could be simplified for such interrupts so that instead of this:
```rust
#[task(binds = EXTI15_10, local = [button, led])]
```

you could do something like this (just pseudo-code; basically anything where you can pick the interrupt in such a way would be nice):
```
#[task(binds = button.interrupt(), local = [button, led])]
```

or, if it isn't possible to do this fully dynamic, then maybe something where you can explicitly say (at compile time) for which pin the external interrupt should be?

note: i've raised this here since i encountered this while testing it with RTIC, but i realise that it might be that the solution for this lies in the HAL crate instead. if so, please let me know and i can raise it there instead.

AfoHT, 2022-11-16:
I'm afraid this is a bit outside the scope of RTIC, and as you conclude it might be more suited for a HAL/PAC, as each hardware does it slightly differently.

It was a while since I worked with STM32s, but be sure to also check the reference manual, together with the datasheet.

RTIC should be happy as long as it is provided a NVIC interrupt-vector, so it should be feasible what you pictured :+1:

closed 2022-11-16 by AfoHT

rursprung, 2022-11-21:
> RTIC should be happy as long as it is provided a NVIC interrupt-vector, so it should be feasible what you pictured 👍

i don't think that this works due the way RTIC is currently generating the code: it expects the `binds` value to be the name of an interrupt which it can copy 1:1 textually, see e.g. here:
https://github.com/rtic-rs/cortex-m-rtic/blob/a79cc083803d126d90d1bb28af61dc6b168a0d41/macros/src/codegen/assertions.rs#L31-L33
or here:
https://github.com/rtic-rs/cortex-m-rtic/blob/a79cc083803d126d90d1bb28af61dc6b168a0d41/macros/src/codegen/hardware_tasks.rs#L42

or am i missing something which would make this work nevertheless?

so while the PAC might offer me a way to get the interrupt name, that still wouldn't be enough.
@AfoHT: would it make sense to re-open this ticket to see if/how RTIC could support something like this? then, in a next step, the PACs could be extended to support this (if an extension is even needed)