rtic-rs/rtic #665: Tasks bound to interrupts can't be fully excluded via #[cfg]

little-arhat, 2022-11-12:
Modifying `cfg-whole-task` example:

```
//! examples/cfg-whole-task.rs

#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

use panic_semihosting as _;

#[rtic::app(device = lm3s6965,  dispatchers = [SSI0, QEI0])]
mod app {
    use cortex_m_semihosting::debug;

    #[shared]
    struct Shared {
        #[cfg(never)]
        unused: u32,
        count: u32,
    }

    #[local]
    struct Local {

    }

    #[init]
    fn init(_: init::Context) -> (Shared, Local, init::Monotonics) {

        (
            Shared {
                #[cfg(never)]
                unused: 1,
                count: 0,
            },
            Local {
            },
            init::Monotonics(),
        )
    }

    #[idle]
    fn idle(_: idle::Context) -> ! {
        debug::exit(debug::EXIT_SUCCESS); // Exit QEMU simulator

        loop {
            cortex_m::asm::nop();
        }
    }

    #[task(binds=UART2, shared = [count])]
    fn foo(mut cx: foo::Context) {
        cx.shared.count.lock(|count| *count += 1);
    }

    // The whole task should disappear,
    // currently still present in the Tasks enum
    #[cfg(never))]
    #[task(binds=UART1, shared = [count])]
    fn foo2(mut cx: foo2::Context) {
        cx.shared.count.lock(|count| *count += 10);
    }
}

```

building it yields:
```
[... 737 lines cut ...]
                you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::interrupt::SSI0,
            );
            #[inline(never)]
            fn __rtic_init_resources<F>(f: F)
            where
                F: FnOnce(),
            {
                f();
            }
            __rtic_init_resources(|| {
                let (shared_resources, local_resources, mut monotonics) =
                    init(init::Context::new(core.into()));
                __rtic_internal_shared_resource_count
                    .get_mut()
                    .write(core::mem::MaybeUninit::new(shared_resources.count));
                rtic::export::interrupt::enable();
            });
            idle(idle::Context::new(&rtic::export::Priority::new(0)))
        }
    }
}

```

As one can see, when no interrupts are used task disappears completely.

little-arhat, 2023-01-17:
Hello! Any updates on this? Or maybe suggestions on how to deal with the root issue (diff configurations)?

David-OConnor, 2023-01-21:
I can confirm this is still an issue. I run into it as well.

korken89, 2023-01-22:
Hi,

The main issue is that cfg support is unfortunately broken in RTIC 1 and getting it to work is very complex.

We're on the path of declaring cfg support a wont-fix for RTIC 1 and revisit the issue in the much simplified codegen of RTIC 2.

AfoHT, 2023-01-22:
The experiment in #691 paved way for getting #692  in without too much fuss, hopefully.

Please have a test run and see if it catches all your needs, might be something I have not been able to test.

It does however remove all references to `foo2` and `foo3` in the generated code.

commit d43c2b64 (2023-01-25): Merge #692

692: CFG: Support HW tasks, cleanup for SW tasks r=korken89 a=AfoHT

Fixes #665
PR #692 (2023-01-25): CFG: Support HW tasks, cleanup for SW tasks
Fixes #665
closed 2023-01-25 by bors[bot]