rtic-rs/rtic #667: Why are panic_halt and shared resources not allowed when compiling for thumbv6?

n8tlarsen, 2022-12-06:
I came across a compile error (__rtic_internal_V6_ERROR) today and traced it back to this snippet from src/export.rs:
```
#[cfg(not(have_basepri))]
pub const fn no_basepri_panic() {
    panic!("Exceptions with shared resources are not allowed when compiling for thumbv6 or thumbv8m.base. Use local resources or `#[lock_free]` shared resources");
}
```
I'm well aware of the limitations for RTIC presented by the Armv6-M architecture (specifically no basepri present), but I'm not following why entering an infinite loop as given by
```
#[panic_handler]
fn panic(_info: &PanicInfo) -> ! {
    loop {
        atomic::compiler_fence(Ordering::SeqCst);
    }
}
```
in [panic_halt](https://github.com/korken89/panic-halt/blob/master/src/lib.rs) should be reason not to allow shared variables. Maybe the error is meant to prevent other panic behaviors, but in that case the blanket error seems rather pedantic. Compiling with panic_abort on nightly works fine, but I'd like to avoid using nightly features in my project if possible. Could someone shed some light on this design decision for me?

n8tlarsen, 2022-12-06:
Or am I confusing the definition of "exceptions"? Is this related to #105?

korken89, 2022-12-07:
Indeed, you are confusing exceptions and panics.

Exceptions are the exception interrupts in the Cortex-M core, and has nothing to do with panics. :)
The issue with exceptions and v6 is that source masking does not work atomically there, so the locking has a race-condition if we were to allow it.
Hence the error you are getting. :)

n8tlarsen, 2022-12-07:
Thanks for getting back to me :)
1. I know the lock used to set PRIMASK when using Armv6-M which is atomic; does it now mask individual interrupts using NVIC_ISER and NVIC_ICER?
2. If the lock were to mask individual interrupts, I'm not following how the non-atomic read, modify, write of NVIC_ISER and NVIC_ICER would cause a data race, since preempting exceptions which modify the NVIC interrupt enable state should return the state when leaving its context.

n8tlarsen, 2022-12-15:
Binding tasks to core exceptions, SysTick, HardFault, SVPend, SVCall, etc result in this error. Core exceptions are not maskable by the NVIC (which handles locking in Armv6m) so sharing data between an exception handler and another task causes the data race. Maybe this error message could be improved to specify something like "core exceptions" to point toward those exceptions which are handled directly by the CPU core, rather than the NVIC.

perlindgren, 2022-12-21:
Once the under the hood docs are done, we could consider pointing the reader to the docs for an in depth explanation.

perlindgren, 2023-01-04:
We close this for now awaiting documentation.

closed 2023-01-04 by perlindgren