rtic-rs/rtic #685: Task reschedule, How to do?

yann77500, 2023-01-11:
Hello, 
I have a question about task (re)schedule. How to schedule a task ‘B’ in order to start 10 secondes (for example) after the last execution of a task ‘A’, as a timeout behaviour. 
I seen example with SpawnHandle and reschedule but I am not able to integrate this solution into my code. 
I tried to store SpawnHandle into Local ressource of task ‘A’ in order to task ‘A’ spawn the task ‘B’ and store the SpawnHandle of task ‘B’. But I have this compilation error  
« cannot move out of `*ctx.local.handler_B` which is behind a mutable reference »

Bellow extraction of program to tried to explain my situation : 

    Local {
	    handler_B : SpawnHandle,
    }

    // INTO TASK A 

	match task_B::spawn_after(10.secs()) {
                    Ok(val) => {
                        ctx.local.handler_B= val;
                    }
                    Err(e) => {
		ctx.local.handler_B.reschedule_after(10.secs())
             
             // Below code tried too 
             //*ctx.local.handler_B.reschedule_after(10.secs())
		}
	}

Certainly there is some information that I have not understand fully, What is wrong ? 
Anyone have a solution for this situation ?

Thank you for your help.

korken89, 2023-01-11:
Hi,

What's missing is that you don't have a way to encode if you have a spawn handle or not.
You need to change `handler_B: SpawnHandle` to `handler_B: Option<SpawnHandle>` so you can move a spawn handle into the storage and then take it out with `ctx.local.handler_B.take().map(|handle| handle.reschedule_after(10.secs()))`.

yann77500, 2023-01-11:
Thanks for your help, with Option encapsulation and take function it's work well.

AfoHT, 2023-01-11:
Awesome! Sounds like you got it working, so I'll close the issue

closed 2023-01-11 by AfoHT