rtic-rs/rtic #693: How to take ownership of a local resource from hardware task

DmitriLyalikov, 2023-01-25:
I am initializing a spi device (rp2040_hal) in my init and exposing it to the local resources. 

My hardware task needs to have access to the transfer trait, which I believe requires ownership. From what I understand,
this task context only has access via mutable reference to the spi peirpheral, therefore giving me the issue:

|         let t = cx.local.spi_dev.transfer(&mut rx_buf, &tx_buf);
   |                                  ^^^^^^^^ method not found in `&mut Spi<Enabled, SPI0, 8>`
'
#[task(binds=SPI0_IRQ, priority=3, local=[spi_dev])]
    fn spi0_irq(cx: spi0_irq::Context) {
        let tx_buf = [1_u8, 2, 3, 4, 5, 6];
        let mut rx_buf = [0_u8; 6];
        let spi_dev = cx.local.spi_dev;
        spi_dev.transfer(&mut rx_buf, &tx_buf); 
    }
'
How can I either take ownership of the spi device in my task or get around this? 

using: 
https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/examples/spi.rs
https://github.com/rust-embedded/embedded-hal/blob/master/embedded-hal/src/spi.rs

AfoHT, 2023-01-25:
Maybe you already had a look at [this example](https://github.com/rtic-rs/rtic-examples/blob/master/rtic_v1/rp-pico_local_initilzd_resources/src/main.rs), although it is using I2C. And as we don't have your code it's hard to test it out/get the full picture of your issue.

AfoHT, 2023-01-25:
Another pointer that might be helpful is https://rtic.rs/1/book/en/by-example/tips_static_lifetimes.html

yann77500, 2023-01-25:
Hello, 
Maybe this solution is to much for your use case, but alternatively you can use shared-bus-rtic https://github.com/ryan-summers/shared-bus-rtic. 
Shared-bus permit to shared multiple devices on peripheral bus like SPI or I2C. It's pretty usefull !

DmitriLyalikov, 2023-01-25:
> Maybe you already had a look at [this example](https://github.com/rtic-rs/rtic-examples/blob/master/rtic_v1/rp-pico_local_initilzd_resources/src/main.rs), although it is using I2C. And as we don't have your code it's hard to test it out/get the full picture of your issue

Hi, @AfoHT Here is my init function that I use to initialize my device. My problem is that init owns the spi peripheral and placing it in the local space simply gives a mutable reference to that bus. I am exploring using the Option<T> .take() method
as seen here: https://users.rust-lang.org/t/rtic-and-using-dma-in-interrupt-handler/62288/2 
which will allow me to take the peripheral into the ownership of the spi task. However, this will require a 'static lifetime, from what i understand, and I wonder if there is a better way

#[rtic::app(device = rp2040_hal::pac, peripherals = true)]
mod app {

    use embedded_hal::digital::v2::OutputPin;
    use embedded_hal::blocking::spi::transfer;
    use embedded_time::duration::Extensions;

    use rp2040_hal as hal;
    use hal::clocks::Clock;
    use hal::pac as pac;
    use fugit::RateExtU32;

    /// External high-speed crystal on the Raspberry Pi Pico board is 12 MHz. Adjust
    /// if your board has a different frequency
    const XTAL_FREQ_HZ: u32 = 12_000_000u32;

    #[shared]
    struct Shared {}

    #[local]
    struct Local {
        spi_dev: rp2040_hal::Spi<hal::spi::Enabled, pac::SPI0, 8>,
    }

    #[init(local = [])]
    fn init(c: init::Context) -> (Shared, Local, init::Monotonics) {
        //*******
        // Initialization of the system clock.
        let mut resets = c.device.RESETS;
        let mut watchdog = hal::watchdog::Watchdog::new(c.device.WATCHDOG);

        // Configure the clocks
        let clocks = hal::clocks::init_clocks_and_plls(
            XTAL_FREQ_HZ,
            c.device.XOSC,
            c.device.CLOCKS,
            c.device.PLL_SYS,
            c.device.PLL_USB,
            &mut resets,
            &mut watchdog,
        )
        .ok()
        .unwrap();

        // The single-cycle I/O block controls our GPIO pins
        let sio = hal::Sio::new(c.device.SIO);

        // Set the pins to their default state
        let pins = hal::gpio::Pins::new(
            c.device.IO_BANK0,
            c.device.PADS_BANK0,
            sio.gpio_bank0,
            &mut resets,
        );

        // These are implicitly used by the spi driver if they are in the correct mode
        let _spi_sclk = pins.gpio6.into_mode::<hal::gpio::FunctionSpi>();
        let _spi_mosi = pins.gpio7.into_mode::<hal::gpio::FunctionSpi>();
        let _spi_miso = pins.gpio4.into_mode::<hal::gpio::FunctionSpi>();
        let spi = hal::Spi::<_, _, 8>::new(c.device.SPI0);
        
        // Exchange the uninitialised SPI driver for an initialised one
        let mut spi_dev = spi.init(
            &mut resets,
            clocks.peripheral_clock.freq(),
            16.MHz(),
            &embedded_hal::spi::MODE_0,
            true,
        );

        //********
        // Return the Shared variables struct, the Local variables struct and the XPTO Monitonics
        //    (Note: Read again the RTIC book in the section of Monotonics timers)
        (
            Shared {},
            Local {
                spi_dev
            },
            init::Monotonics(),
        )
    }

DmitriLyalikov, 2023-01-25:
> Hello, Maybe this solution is to much for your use case, but alternatively you can use shared-bus-rtic https://github.com/ryan-summers/shared-bus-rtic. Shared-bus permit to shared multiple devices on peripheral bus like SPI or I2C. It's pretty usefull !

> Hello, Maybe this solution is to much for your use case, but alternatively you can use shared-bus-rtic https://github.com/ryan-summers/shared-bus-rtic. Shared-bus permit to shared multiple devices on peripheral bus like SPI or I2C. It's pretty usefull !

Hi @yann77500 . I will try this and let you know. Is this an RTIC  implementation that simply supports this bus sharing feature?

yann77500, 2023-01-25:
> Hi @yann77500 . I will try this and let you know. Is this an RTIC implementation that simply supports this bus sharing feature?

Yes, this crate only add sharing feature for I2C and SPI bus between multiple drivers and multiple task. I don’t think in your case you need to shared SPI over multiple devices, so may be shared-bus-rtic is too much. But for now, is the only solution that I know to share bus.
If you don’t need to share on multiple driver, the solution given by @AfoHT  should be useful.  

I tried shared-bus-rtic it on cortex M4 MCU without any problem. 
Becarful with example given on shared-bus-rtic, I think they are based on previous RTIC version. 
You need to adapt by putting shared_bus_resources into your local struct instead of ressource structure

DmitriLyalikov, 2023-01-26:
Guys, this issue is resolved. After poking around with this and finding a solution based on both the I2C example and using the core::option::Option type, I realized I just needed to use the trait
 
embedded_hal::blocking::spi::Transfer; 

which can be done without ownership just a &mut, basically solvng my original problem. Thanks for your help.
Now something like this works just fine:
    #[inline(never)]
    #[link_section = ".data.bar"]
    #[task(binds=SPI0_IRQ, priority=3, local=[spi_dev])]
    fn spi0_irq(cx: spi0_irq::Context) {
        let mut tx_buf = [1_u16, 2, 3, 4, 5, 6];
        let mut _rx_buf = [0_u16; 6];
        let _t = cx.local.spi_dev.transfer(&mut tx_buf);
    }

thanks for your help lads

closed 2023-01-26 by DmitriLyalikov