rtic-rs/rtic #695: rp2040 SPI0_IRQ appears to not be asserting

DmitriLyalikov, 2023-02-11:
Hi everyone,

I am working on a creating a spi slave on the RP2040 that will respond via the SPI0_IRQ assigned task. 
This uses the RP2040 Hal and pi-pico BSP

Here is my setup in the #init function : 
`       let _spi_sclk = pins.gpio18.into_mode::<hal::gpio::FunctionSpi>();
        let _spi_mosi = pins.gpio17.into_mode::<hal::gpio::FunctionSpi>();
        let _spi_miso = pins.gpio16.into_mode::<hal::gpio::FunctionSpi>();
        let _spi_cs = pins.gpio19.into_mode::<hal::gpio::FunctionSpi>();
        let spi = hal::Spi::<_, _, 8>::new(c.device.SPI0);
        // Exchange the uninitialized spi device for an enabled slave
        let mut spi_dev = spi.init_slave(&mut resets, &embedded_hal::spi::MODE_0);`

and here is my task bound the SPI0_IRQ

`    // Task that binds to the SPI0 IRQ and handles requests. This will execute from RAM
    // This takes a mutable reference to the SPI bus writes immediately from tx_buffer while reading 
    // into the rx_buffer
    #[inline(never)]
    #[link_section = ".data.bar"] // Execute from IRAM
    #[task(binds=SPI0_IRQ, priority=4, local=[spi_dev, spi_tx_consumer], shared = [serial])]
    fn spi0_irq(cx: spi0_irq::Context) {
        let mut tx_buf = [5u8; 18];
        // Write/Read words back to slave. Received words will replace contents in tx_buf
        if let Some(mut tx_buf) = cx.local.spi_tx_consumer.dequeue(){}
        
        cx.local.spi_dev.transfer(&mut tx_buf).unwrap();
        // Received words, Now build our HostRequest
        match HostRequest::new().build_from_8bit_spi(&tx_buf) {
            Ok(hr) => { // Host Request is clean and ready to be sent out
                send_out::spawn(hr);
            }
            Err(err) => { // Print the error to serial port if Host Request is invalid
            }
        }
    }`

My problem is that when asserting chip select from my spi master, I don't see this task invoke. Which makes me think that either I need to configure the interrupt.  Could there be anything in RTIC configuration that is not on by default, like could the SPI interrupt not be enabled? 

Here is my captured output. 
[image]

DmitriLyalikov, 2023-02-12:
Update: I am led to believe that the SPI0_IRQ is not enabled in the NVIC. 
I would like to test this and write manually to the NVIC ISER register to read if it is enabled, and if not, enable it and continue the debugging procedure. 

However, I am not sure how to do this safely in my RTIC application. when doing this: 
`use rtic::export::NVIC;`

My entire application becomes unsafe. 
What is the best way to go about writing/reading to the Cortex-M0+ NVIC registers?

korken89, 2023-04-15:
Hi, were you able to get this resolved?

korken89, 2023-07-27:
If this issue persists, feel free to reopen this issue!

closed 2023-07-27 by korken89