rtic-rs/rtic #700: Support for AVR boards such as the atmega32u4?

qwelyt, 2023-03-08:
I've used rtic for projects using rp2040 and really liked the experience. Now I'm working on a project that uses the 32u4 chip and was wondering if it's possible to use rtic with this chip as well?

As far as I can tell from the datasheets for the 32u4 ( https://ww1.microchip.com/downloads/en/devicedoc/atmel-7766-8-bit-avr-atmega16u4-32u4_datasheet.pdf ) the chip itself should be able to provide what's needed (supports interrupts etc). But I'm no good at low level code really so not really sure about it.

So my question is: Is it possible to use rtic with the 32u4 chip? The crate name suggests otherwise, but perhaps there is work being done to support the AVR-family of chips?

Best regards

AfoHT, 2023-03-08:
There is some effort, but I am not up to date with the current state of Rust + AVR

https://github.com/avr-rust

The current RTIC v1, as you noted based on the crate name, does not support anything but Cortex-M.
With the recent work of decoupling architecture and RTIC, driven by RISC-V interest I hope that the barrier to also doing something for AVR is lowered.

But as of now there is nobody working on AVR to my knowledge, would be thrilled to hear if there are!

qwelyt, 2023-03-08:
The state of rust + avr is steadily improving. With works like https://github.com/agausmann/atmega-usbd coming along nicely.

I'm fairly inexperienced with Rust but I might take a stab at trying to make rtic-avr a thing. I'll see what I can come up with. But if anyone know of someone else already working on it please let me know!

korken89, 2023-04-15:
Closing as under development.

closed 2023-04-15 by korken89

NailAgliev, 2024-03-04:
Hi there! Would like an update on the state of the implementation of AVR platform support. I would like to take part in this feature's implementation.

AfoHT, 2024-03-06:
With #815 the "different platform" support gets a major lift. With this as the outset I think this is open for any attempts :)

I don't know of any current ongoing AVR work