rtic-rs/rtic #708: how to access peripherals in pre_init?

cosmikwolf, 2023-03-16:
when I attempt to add the cx context to a pre_init function as indicated as possible in: 

https://rtic.rs/1/book/en/by-example/app_init.html

I receive the following error:

```
error: `#[pre_init]` function must have signature `unsafe fn()`
```

The piece of documentation I am referring to is: 

> The init and optional pre-init tasks runs with interrupts disabled and have exclusive access to Cortex-M (the bare_metal::CriticalSection token is available as cs).

My goal here is to try and access the last reset reason to see if it was a software reset. 

Am I trying to do something impossible here?

Yandrik, 2023-03-19:
Does your function look something like this?
```rust
#[pre_init]
unsafe fn before_main() {
    // do something here
}
```
The pre_init function must be unsafe, according to the [API docs here](https://docs.rs/cortex-m-rt/latest/cortex_m_rt/attr.pre_init.html), because it's called before memory is initialized. Make sure to read the section about undefined behavior, but other than that accessing peripherals should be possible.
Though, if possible, doing this at the start of `init` is a far better idea.

korken89, 2023-04-07:
Hi,

That was unfortunately worded in the documentation, it's not supported to add pre init in RTIC.
One needs to use the cortex_m_rt provided pre_init.

After that, at access peripherals, you need to conjure them using unsafe.

korken89, 2023-04-15:
This should be resolved, closing.

closed 2023-04-15 by korken89