rtic-rs/rtic #713: Ownership difference between tuple locking and nested locking

nkrackow, 2023-03-22:
I have come across a pattern that seems strange to me and that I have not fund any mention of in the documentation or anywhere else:

When I lock several rtic resources in a tuple at once it takes ownership of the resources, when I lock a single resource (or the same ones in a nested lock) it does not.

Minimum example:
```rust
(c.shared.resource0, c.shared.resource1).lock(|r0, r1| {
   r0.foo();
   r1.foo();
});

c.shared.resource0.lock(|r0| r0.foo());
``` 
will fail because I try to use the moved `resource0` in the last line. It was moved in the tuple lock.

```rust
c.shared.resource0.lock(|r0| {
   c.shared.resource1.lock(|r1| {
      r0.foo();
      r1.foo();
   });
});

c.shared.resource0.lock(|r0| r0.foo());
``` 
will compile just fine.

Is this intentional? Or am I missing something?

korken89, 2023-03-22:
Hi,

This is Rust behavior of tuples, they move whatever is place in them.
To the the same behavior as int the nested lock case you need to do `(&mut r0, &mut r1).lock( .. ` :)

nkrackow, 2023-03-23:
Ah, I see. Thanks a lot for clarifying!

closed 2023-03-23 by nkrackow